arduino/arduino-ide #2172: Arduino V2 IDE failed to launch Open OCD  GDB server.

post2pa, 2023-08-17:
### Describe the problem

Using IDE v2 and a simple example sketch, on trying to enter debug mode the following messages arise: 

Waiting for gdb server to start...[2023-08-17T17:08:46.386Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\openOCD\\OpenOCD-20230712-0.12.0\\bin" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Arduino_v2\\Sketchbooks\\oled\\ssd1306_128x32_i2c\\ssd1306_128x64_i2c" -f "C:/Arduinov2.1.1/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f "C:\\Users\\post2\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.8.13/variants/arduino_zero/openocd_scripts/arduino_zero.cfg"

and:

Failed to launch OpenOCD GDB Server. error spawn C:\openOCD\OpenOCD-20230712-0.12.0\bin ENOENT

This is a fresh install of Arduino V2.1.1 and OpenOCD. 

The target board is an Arduino Zero (with embedded EDBG) and using the "programming usb" connected to my windows pc (64bit). 

The example sketch uploads and executes no problem. 

Please can anyone suggest a solution, thank you. 

Regards, Paul.   

### To reproduce

  Just compile, upload, enter debug...

### Expected behavior

enter debug mode automatically.

### Arduino IDE version

v2.1.1

### Operating system

Windows

### Operating system version

Windows 11 Pro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-08-18:
Thanks for your interest in this project. This issue tracker is for bugs and feature requests and does not provide support. Please use the forum instead: https://forum.arduino.cc/c/software/ide-2-x/93/; it's more appropriate.

I tried the debugger with my Arduino Zero board on Windows, and it works as expected when I follow the [guide](https://docs.arduino.cc/tutorials/zero/debugging-with-zero).

closed 2023-08-18 by kittaakos

post2pa, 2023-08-18:
Thank you for your reply, much appreciated.
Regards, Paul.