arduino/arduino-ide #2173: checksums for downloads please

jamcinnes2, 2023-08-18:
### Describe the problem

Can you post checksums for the releases? Somewhere secure https.

### To reproduce

no checksums

### Expected behavior

want checksums

### Arduino IDE version

2.1.1

### Operating system

Linux

### Operating system version

6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Dissineer, 2024-05-14:
SHA-512 for **arduino-ide_2.3.2_Linux_64bit.AppImage** is 02f09d1bbe4a4982143bd23e01c868b95375eda6edfb85d40a66dd1e6dbc69fa96247680809671c3274b5e621d3a5d08f6cc7192239d2a37957c2bcd36f5c421

per1234, 2025-04-12:
Thanks for your suggestion @jamcinnes2. Even though your issue was made first, I see we now have an additional issue that contains a bit more information: https://github.com/arduino/arduino-ide/issues/2648

It is best to have only a single issue for each distinct subject, so I'm going to close this one in favor of the other.

closed 2025-04-12 by per1234