arduino/arduino-ide #2175: Firmware Updater fails when target port has multiple identifications

per1234, 2023-08-18:
### Describe the problem

Arduino boards platforms may associate a board definition with [port identification properties](https://arduino.github.io/arduino-cli/dev/pluggable-discovery-specification/#board-identification). Arduino IDE identifies a port as being that board when the port has these properties.

The platform framework doesn't provide any mechanism for restricting the association of a given set of identification properties to a single board definition, so it is possible (and even inherent to some use cases) for a port to be identified with multiple board definitions.

🐛 If the port of the target board is identified with multiple board definitions, the Firmware Updater fails.

### To reproduce

#### Equipment

One of the boards supported by the "**WiFi101 / WiFiNINA Firmware Updater**" feature:

- [**UNO R4 WiFi**](https://docs.arduino.cc/hardware/uno-r4-wifi)
- [**Nano 33 IoT**](https://docs.arduino.cc/hardware/nano-33-iot)
- [**MKR WiFi 1010**](https://docs.arduino.cc/hardware/mkr-wifi-1010)
- [**MKR 1000 WiFi**](https://docs.arduino.cc/hardware/mkr-1000-wifi)
- [**Nano RP2040 Connect**](https://docs.arduino.cc/hardware/nano-rp2040-connect)
- [**Uno WiFi Rev2**](https://docs.arduino.cc/hardware/uno-wifi-rev2)

#### Steps

1. Select **File > Quit** (or **Arduino IDE > Quit Arduino IDE** for macOS users) from the Arduino IDE menus if it is running.
1. Click the following link to download the dummy boards platform used to efficiently create the conditions required to produce the fault:
   [hardware.zip](https://github.com/arduino/arduino-ide/files/12380797/hardware.zip)
1. Unzip the downloaded file to your sketchbook folder.
   The folder structure should look like this:
   ```text
   <sketchbook location>/
   ├── hardware/
   │   └── fwuploader-board-dummies/
   │       └── foo/
   │           └── boards.txt
   [...]
   ```
1. Start Arduino IDE.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
1. Select "**Developer (Arduino): Clear the Board and Port Selection**" from the Command Palette menu.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
1. Select "**Developer (Arduino): Clear the Board List History**" from the Command Palette menu.
   **ⓘ** This and the three steps before are not required to reproduce the fault that is the primary subject of this report. They are included in order to demonstrate a related fault associated with the conditions of using "**WiFi101 / WiFiNINA Firmware Updater**" when there are multiple identifications of the target port.
1. Connect the firmware update target board to your computer with a USB cable if it is not already.
1. Select **Tools > WiFi101 / WiFiNINA Firmware Updater** from the Arduino IDE menus.
   The "**WiFi101 / WiFiNINA Firmware Updater**" dialog will open.
1. Attempt to select the target board from the "**Select Board**" menu in the "**WiFi101 / WiFiNINA Firmware Updater**" dialog.
   🐛 The menu doesn't open.
1. Close the "**WiFi101 / WiFiNINA Firmware Updater**" dialog.
1. Select the port of the target board from the Board Selector menu on the Arduino IDE toolbar.
   The "**Select Other Board and Port**" dialog will open.
1. Select the board name that does not have a "(dummy)" suffix from the "**BOARDS**" menu.
1. Click the "**OK**" button.
1. Select **Tools > WiFi101 / WiFiNINA Firmware Updater** from the Arduino IDE menus.
1. Select the target board from the "**Select Board**" menu in the "**WiFi101 / WiFiNINA Firmware Updater**" dialog.
1. Click the "**CHECK UPDATES**" button.
1. Click the "**INSTALL**" button.

🐛 The update process immediately fails:
> **⚠ Installation failed. Please try again.**

#### Clean up

1. Delete the folder of the dummy platform at `hardware/fwuploader-board-dummies`

### Expected behavior

It is possible to update firmware on target boards for which the port has multiple identifications.

### Arduino IDE version

69ae38e

### Operating system

Windows

### Operating system version

11

### Additional context

I bisected the regression to 69ae38e (neither of the faults occur when using the build from the previous commit 9a6a457bc44fb16f646b9e61a9e8b336b3666412 under the same conditions).

---

The minimal "dummy" boards platform used to produce the required conditions in the demo is not anything a real user would have installed, but the fault also occurs in the more common (but more work to to reproduce) conditions of having an additional real boards platform installed that associates a board definition with the identification properties of the target board's port. A real world example of this would be having both a release version and a development version of a platform installed at the same time, as is common for platform developers and beta testers.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-08-19:
> 11\. Attempt to select the target board from the "**Select Board**" menu in the "**WiFi101 / WiFiNINA Firmware Updater**" dialog.
>      🐛 The menu doesn't open.

What do you expect to see in the `<select>` before confirming the board between steps `(12)` and `(15)`? Do you expect to see two board items? This would imply showing 27 different ESP32 boards on the same port in some corner cases. Thank you

per1234, 2023-08-19:
>Do you expect to see two board items?

The menu should only offer the boards that are supported by Arduino Firmware Uploader:

```text
$ ./arduino-fwuploader.exe firmware list
Board                       FQBN                                Module     Version
Arduino MKR1000             arduino:samd:mkr1000                WINC1500   19.4.4 
Arduino MKR1000             arduino:samd:mkr1000                WINC1500   19.5.2 
Arduino MKR1000             arduino:samd:mkr1000                WINC1500   19.5.4 
Arduino MKR1000             arduino:samd:mkr1000                WINC1500 ✔ 19.6.1 
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.0.0  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.1.0  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.2.1  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.2.2  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.2.3  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.2.4  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.3.0  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.0  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.1  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.2  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.3  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.4  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.5  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.6  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.7  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA       1.4.8  
Arduino MKR WiFi 1010       arduino:samd:mkrwifi1010            NINA     ✔ 1.5.0  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.0.0  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.1.0  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.2.1  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.2.2  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.2.3  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.2.4  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.3.0  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.0  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.1  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.2  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.3  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.4  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.5  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.6  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.7  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA       1.4.8  
Arduino NANO 33 IoT         arduino:samd:nano_33_iot            NINA     ✔ 1.5.0  
Arduino MKR Vidor 4000      arduino:samd:mkrvidor4000           NINA       1.0.0  
Arduino MKR Vidor 4000      arduino:samd:mkrvidor4000           NINA     ✔ 1.1.0  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.2.1  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.2.2  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.2.3  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.2.4  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.3.0  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.0  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.1  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.2  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.3  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.4  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.5  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.6  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.7  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA       1.4.8  
Arduino Uno WiFi Rev2       arduino:megaavr:uno2018             NINA     ✔ 1.5.0  
Arduino Nano RP2040 Connect arduino:mbed_nano:nanorp2040connect NINA       1.4.5  
Arduino Nano RP2040 Connect arduino:mbed_nano:nanorp2040connect NINA       1.4.6  
Arduino Nano RP2040 Connect arduino:mbed_nano:nanorp2040connect NINA       1.4.7  
Arduino Nano RP2040 Connect arduino:mbed_nano:nanorp2040connect NINA       1.4.8  
Arduino Nano RP2040 Connect arduino:mbed_nano:nanorp2040connect NINA     ✔ 1.5.0  
Arduino UNO R4 WiFi         arduino:renesas_uno:unor4wifi       ESP32-S3   0.1.0  
Arduino UNO R4 WiFi         arduino:renesas_uno:unor4wifi       ESP32-S3   0.2.0  
Arduino UNO R4 WiFi         arduino:renesas_uno:unor4wifi       ESP32-S3 ✔ 0.2.1  
```

[... cut ...] ported multiple boards that shared the same port identification properties, as long as the boards supported by tool continues to be curated by Arduino, there would never be a case where there was a problematic number of supported boards for a given port.

In the demo I provided, there are two general identifications for the port (e.g., `arduino:renesas_uno:unor4wifi`, `fwuploader-board-dummies:foo:unor4wifi`), but only one of the identifications is for a board supported by the Firmware Uploader.

commit db01efea (2023-08-20): fix: expand boards if available on detected port

moved the board inference logic from UI to  model

Closes #2175
closed 2023-08-20 by kittaakos