arduino/arduino-ide #2209: macOS Arduino IDE2.2.1 double clicking any .ino file won't open IDE window

pacastro, 2023-08-31:
### Describe the problem

After upgrading from IDE 2.1.1 to 2.2.1, double clicking on any .ino file won't open IDE window anymore. Arduino IDE icon appears on the bottom dock, so it seems to be running but no IDE window with the .ino file open appears.

Opening Arduino IDE app directly from launchpad works OK, and it opens the last .ino file on the IDE window.  Once you have the IDE window opened this way, further double clicking any .ino file works OK and opens a new IDE window.

The problem exists when no IDE window is previously opened, then double clicking any .ino file won't open an IDE window.

### To reproduce

With no previous instance of IDE2.2.1 running, double click on any .ino file. No IDE window opens.

### Expected behavior

With no previous instance of IDE2.2.1 running, after double clicking any .ino file an IDE window should open with the file clicked opened in it.

### Arduino IDE version

2.2.1

### Operating system

macOS

### Operating system version

Ventura 13.4

### Additional context

arduino-ide_2.2.1_macOS_arm64

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-08-31:
Hi @pacastro. Thanks for your report. Arduino IDE automatically generates log files while it is running. I'm going to ask you to provide the latest log file from your computer. It might contain information that will help us to identify the problem.

Please do this:

1. Attempt to open a `.ino` file in Arduino IDE, just as you did before.
1. Wait some time to be sure any activity by the triggered non-functional Arduino IDE process has finished.
1. Open the issue page here on GitHub:
   https://github.com/arduino/arduino-ide/issues/2209#issue-comment-box
1. Click the "**Attach files by dragging & dropping, selecting or pasting them**" at the bottom of the "**Leave a comment**" field.
1. Select the latest `.log` file from this folder (the files are named according to the date):
   ```text
   /Users/<username>/Library/Logs/Arduino IDE/
   ```
   (where `<username>` is your macOS username)
1. Click the "**Open**" button to attach the file to the comment.
1. You will now see an "**Uploading your files... (1/1)**" status indicator at the bottom of the "**Leave a comment**" field. Wait until that disappears.
1. Click the "**Comment**" button to publish the comment.

---

Please let me know if you have any questions or problems while following those instructions.

pacastro, 2023-08-31:
[2023-08-31_log.log](https://github.com/arduino/arduino-ide/files/12489686/2023-08-31_log.log)

per1234, 2023-09-28:
Thanks @pacastro. I am able to reproduce the fault on my macOS machine.

I'll provide my own logs that were generated from starting the IDE by double clicking the `.ino` file of a valid sketch:

[2023-09-27_log.log](https://github.com/arduino/arduino-ide/files/12746210/2023-09-27_log.log)

They will be easier to read through because they only contain the content from a single launch (rather than the accumulation of content from previous sessions as the logs usually contain).

The fault does not occur if I start the IDE from the terminal, passing the path to the `.ino` file as a command line argument.

ulrichloose, 2023-10-15:
The same on MacOS:

Since 2.1.1 double clicking on a .ide file opens the ide but not the .ide file itself. If die IDE is already running, then a double click on an .ide file opens. In version 2.1.1 the ide starts AND opens the double clicked .ide file.

commit b17607c8 (2023-11-22): fix: can open IDE from an ino file

Use a fallback frontend app config when starting IDE2 from an ino file
(from Explorer, Finder, etc.) and the app config is not yet set.

Closes #2209
PR #2288 (2024-01-15): fix: can open IDE2 from an ino file
### Motivation

Can open IDE2 from an ino file.

### Change description

Use a fallback frontend app config when starting IDE2 from an ino file
(from Explorer, Finder, etc.) and the app config is not yet set.

### Other information

See the [Beta Testing Guide](https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md) on how to verify the changes.

Preconditions:
 - make sure the tester build is your default app to open ino files (On macOS, I had to wipe all previous IDE2 versions)

Steps to verify:
 - download the tester build [... cut]
commit 3eef857b (2024-01-15): fix: can open IDE from an ino file

Use a fallback frontend app config when starting IDE2 from an ino file
(from Explorer, Finder, etc.) and the app config is not yet set.

Closes #2209
closed 2024-01-15 by kittaakos