arduino/arduino-ide #2213: Missing translation for "save" in German

Sinclair-ZX81, 2023-09-03:
### Describe the problem

Arduino IDE 2.2.1:
(also tested with 2.2.2-nightly-20230903)

[image]

### To reproduce

Just open the IDE Files menu

### Expected behavior

 The German word for **Save** is **Speichern**

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

Windows 10 x64

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-09-03:
Thanks for reporting this @Sinclair-ZX81.

I confirm that there was a regression in the localization of the **File > Save** menu item. I bisected the regression to 192aac5a810d8ff1da788e15156009095999b656, where the language packs were bumped from 1.53.2 to 1.70.0.

The localization data key for the menu item is set here:

https://github.com/arduino/arduino-ide/blob/fe53b8e0d048ef9d135381383204cd68afa69e7b/arduino-ide-extension/src/browser/contributions/save-sketch.ts#L28

The data is here in version 1.53.2 of the language pack:

https://github.com/microsoft/vscode-loc/blob/295dfb35491adc41df4b89e0d839086131a34257/i18n/vscode-language-pack-de/translations/main.i18n.json#L5000

```json
		"vs/workbench/contrib/files/browser/fileCommands": {
			"saveAs": "Speichern unter...",
			"save": "Speichern",
```

Without explanation of the reason for the breaking change, the data was moved in the language packs from `vs/workbench/contrib/files/browser/fileCommands/save` to `vs/workbench/contrib/files/browser/fileConstants/save` by https://github.com/microsoft/vscode-loc/commit/e36740c0d1c65bac2391909fe2eb4b366800ea36:

https://github.com/microsoft/vscode-loc/blob/a51eba0bb6e14e9b26cb7b5fed1aeca520552010/i18n/vscode-language-pack-de/translations/main.i18n.json#L6189

```json
		"vs/workbench/contrib/files/browser/fileCommands": {
			"discard": "Verwerfen",
			"genericRevertError": "Fehler beim Zurücksetzen von '{0}': {1}",
			"genericSaveError": "Fehler beim Speichern von \"{0}\": {1}",
			"modifiedLabel": "{0} (in Datei) ↔ {1}",
			"retry": "Wiederholen"
		},
		"vs/workbench/contrib/files/browser/fileConstants": {
			"newUntitledFile": "Neue unbenannte Datei",
			"removeFolderFromWorkspace": "Ordner aus dem Arbeitsbereich entfernen",
			"save": "Speichern",
```

danik292, 2023-09-07:
I fix it

danik292, 2023-09-07:
@Sinclair-ZX81 after commiting PR check nightly build

Sinclair-ZX81, 2023-09-07:
Will do that - thanks