arduino/arduino-ide #2214: Feature Request: Add TensorflowJS-Converter to the installation of IDE2x

hpssjellis, 2023-09-03:
### Describe the request

@kittaakos @per1234 

**Feature Request: Add [TensorflowJS-Converter](**https://www.tensorflow.org/js/guide/conversion**) to the installation of IDE2x**

TensorflowJS allows simple webpage based creation of small machine learning models that would be perfect for microprocessors. These models can then easily be command line converted from Tensorflow format to Keras format and then to TFLITE format, which can then be installed on Arduinos like the PortentaH7.

The problem is that many Javascript programmers can't or don't have the interest in installing a Node/Python/command line package. If the [tensorflow-Converter](https://www.npmjs.com/package/@tensorflow/tfjs-converter) package was installed with the IDE2 then this ability could be completed with only the Arduino IDE2, a huge improvement.

I can provide more information if this is even possible with the IDE2, please reply if slightly interested.

### Describe the current behavior

Currently makers have to load all their private data onto cloud sites like EdgeImpulse to easily create a machine learning model. This behavior is efficient and the best way to make a tinyML model but Makers could make these models on their own devices.

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

22621.2134

### Additional context

Here is an example of using the Arduino Nano33BleSense to make a complete machine learning model on the web and to test the results, but the steps to convert this model to be installed on an Arduino need the [tensorflowJS-converter](https://www.tensorflow.org/js/guide/conversion).

https://hpssjellis.github.io/tinyMLjs/public/acceleration/a00-best-acceleration.html

This model has been shown to several Arduino employees at the ICTP conference in Italy.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-09-04:
Thanks for your suggestion @hpssjellis. Arduino IDE is a tool for developing Arduino sketches, so this would be out of scope for the Arduino IDE project.

If you think integrating such a converter into Arduino IDE would be useful, you might be interested to know that it is possible for the community to add additional capabilities to Arduino IDE through VS Code extensions. Creating a TensorFlow model converter extension might be a fun project for you or another community member to take on.

Arduino IDE's VS Code extension support has recently been enhanced significantly, as announced here:

<https://forum.arduino.cc/t/arduino-ide-2-2-0-is-now-available/1163477#expose-arduino-state-data-for-use-by-vs-code-extensions-11>

closed 2023-09-04 by per1234

hpssjellis, 2023-09-04:
> Arduino IDE is a tool for developing Arduino sketches, so this would be out of scope for the Arduino IDE project.

Respectfully @per1234 , I disagree with the above statement. Machine learning is a part of Arduino Sketches and the converter would allow Makers to directly program Arduino sketches without being tied to potentially expensive cloud services.

All the 3 year old code on this page [here](https://github.com/hpssjellis/my-examples-for-the-arduino-portentaH7/tree/master/m09-Tensoflow) are pure Arduino sketches using machine learning that Arduino C/C++ coding Makers do not have assess to without loading Python and all it's dependencies. If Arduino is moving towards the massive field of ML, these tools would be very helpful.

The VS extension manager idea is very interesting but I typically have my students use the Arduino IDE so that I don't have to teach them the VS method of coding

@mbanzi @dcuartielles 
I will close this request but do think you should review it again as Arduino gets more involved with the field of Machine Learning. To make machine learning code from simple static webpages and upload that code directly onto Arduino Sketches using the Arduino IDE would be very useful.