arduino/arduino-ide #2218: Ping Timeout error with Latest Version 2.2.1

ve3sjk, 2023-09-05:
### Describe the problem

I am seeing random "ping timeout" errors while compiling with the latest update. I have also seen random issues with the com port not being recognized, when it does this the "board" section in the menu is missing. Not really sure what it could be pinging i do not have any internet related functions, all the files are local. There are also times when the compile never completes as well and it requires a shutdown and restart. The arrow remains greyed out, but the compile complete message never happens.

### To reproduce

Issue Verifly or Upload command

### Expected behavior

Compile to complete or error message

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

StevenAMC, 2023-09-08:
I have the same error message.
**Arduino IDE version**
2.2.1
**Operating system**
Ubuntu 22.04
**Additional context**
No response

per1234, 2023-09-24:
Hi @ve3sjk @StevenAMC @scottbob09. Thanks for your report.

There is nowhere near enough information here for us to initiate an investigation.

Please provide detailed instructions I can follow to reproduce the problem.

These should include the following information:

- Which board do you have selected in Arduino IDE?
- Where exactly are you seeing the "ping timeout" errors?

---

Please refrain from replying unless you have significant and unique information to add. If you only want to express your support for the PR or a comment in the thread, use the ["Reactions" feature](https://github.blog/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to add a "👍", etc. "Me too" comments are not allowed here.

Everyone who watches the repository receives a notification whenever any reply is made on any of the issues and PRs. It is important for the developers and maintainers of the project to watch the repository to make sure we see the valuable feedback from the contributors. But if the repository becomes the source of a large number of low value notifications we must either waste our time sorting through them instead of doing productive work, or else unwatch the repository.

If a thread here become the source of low quality noise, it will be locked. That might prevent others from adding actually valuable information, and thus hamper the resolution. So pointless comments harm the users in addition to the project maintainers.

Unstructured conversations about Arduino-related subjects are always welcome over on [Arduino Forum](https://forum.arduino.cc/).

furynick, 2023-09-28:
Same problem here when compiling on slow computers or large projects, IDE tells it's in offline mode and build process is interrupted.
[image]

[image]

furynick, 2023-09-28:
In addition please find debug info from IDE in reply to [ptillisch on arduino forum](https://forum.arduino.cc/t/compile-fails-with-esp32s-weird-message-ping-timeout/1172837/6)

IDE hangs while compiling, "ping timeout" popup can be dismissed but there's no way to even close the IDE.
After some time debug data appears but Powershell buffer, even with 9999 lines don't  allow to store every output so I had to redirect log to files.
[Arduinologs.zip](https://github.com/arduino/arduino-ide/files/12747712/Arduinologs.zip)

MauricioEmilio, 2023-10-09:
My CPU also running as hell... =/

ednieuw, 2023-10-24:
Same problem with esp32 boards like ESP32C3, ESP32S3 and ESP32 RRGO T-watch. 
in The folder c:\Users\ednie\AppData\Local\Temp\  one can find the temporary sketch precompile data
When comparing it with a sketch from the IDE1.8.19 I see a part of the files the IDE2.2.1 stores are stored in another folder named: c:\Users\ednie\AppData\Local\Temp\arduino\sketches\

 The "ping timeout" occurs when the line: 
Using library EEPROM at version 2.0.0 in folder: :\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\EEPROM 
is written.

in the Arduino 1.8.19 the last lines are:

Using library EEPROM at version 2.0.0 in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\EEPROM 
"C:\\Users\\ednie\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\ednie\\AppData\\Local\\Temp\\arduino_build_955993/TTGO-wordClock-V001.ino.elf"
Sketch uses 438129 bytes (6%) of program storage space. Maximum is 6553600 bytes.
Global variables use 24296 bytes (0%) of dynamic memory, leaving 4497688 bytes for local variables. Maximum is 4521984 bytes.

It look like that "xtensa-esp32-elf-size.exe" is not executed properly. 
Maybe part on the necessary files are stored in the wrong folder?

Before the ping out problem  I had the compiler error comment: before after compiling : "connection got disposed"
See #1587

fjmsouza, 2023-12-05:
me too: board ESP32 version 2.0.14
Then, @ednieuw , could you share the steps to fix this in IDE 2.2.1?

ednieuw, 2023-12-05:
I just upgraded the IDE2 to 2.2.1 and ESP32 board to 2.0.14. 
Windows 11 is also up to date.

fjmsouza, 2023-12-06:
Thank you for your attention, @ednieuw 
To me, no way. Same settings, but in Ubuntu 20.04.6 LTS.

ednieuw, 2023-12-07:
Ping timeout occurred again.
I changed from the <Adafruit_ILI9341.h> to  <TFT_eSPI.h> in my program

It stops here: Using library SPIFFS at version 2.0.0 in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\SPIFFS 

Also got the error while trying to save: "Could not save the sketch. Please copy your unsaved work into your favorite text editor, and restart the IDE."
There is also an error in the bottom (left) that says Offline and can not connect to backend. After a minute or so Offline disappears and you can save and exit the IDE
When changing back to the Adafruit library compilation goes well and ends with

Using library Wire at version 2.0.0 in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\Wire 
Using library SPI at version 2.0.0 in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\libraries\SPI 
"C:\\Users\\ednie\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\ednie\\AppData\\Local\\Temp\\arduino\\sketches\\836AB0D4CB6791DC47D8465B9164231B/ESP322432S028_WordClockV002.ino.elf"
Sketch uses 1215513 bytes (61%) of program storage space. Maximum is 1966080 bytes.
Global variables use 59464 bytes (18%) of dynamic memory, leaving 268216 bytes for local variables. Maximum is 327680 bytes.

jgregoire, 2024-01-11:
Arduino 2.2.1, my build terminates at

`Using library SPI at version 2.0.0 in folder: C:\Users\jfg\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\SPI `

With "Ping Timeout" and no additional information.

Otaremwa, 2024-04-27:
I'm ecountering the same "ping timeout" error.

#include<WiFi.h>
#include<Firebase_ESP_Client.h>
#include "addons/TokenHelper.h"
#include "addons/RTDBHelper.h"

#define WIFI_SSID ""
#define WIFI_PASSWORD ""
#define API_KEY "AIzaSyDdQVqDbBd8obG1DIhcuPUChEyDA4Srr4c"
#define DATABASE_URL "grain-monitoring-system-default-rtdb.firebaseio.com/"

#define LED1_PIN 
#define LED2_PIN
#define LDR_PIN
#define PWMChannel 0

const int freq =5000;
const int resolution = 8;

FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
bool signupOK = false;
int ldrData = 0;
float voltage = 0.0;
bool ledStatus = false;

void setup() {
  Serial.begin(115300);
  Wifi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connection to Wi-Fi");
  while(WiFi.status( !WL_CONNECTED){
    Serial.print(""); delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  config.api_key = API_KEY;
  config.database_url = DATABASE_URL;
  if(Firebase.signUp(&config, &auth, "", "")){
    Serial.println("signUp OK");
    signupOK = true;
  }else{
    Serial.printf("%s\n", config.signer.signupError.message.c_str());
  }
  config.token_status_callback = tokenStatusCallback;
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
}

void loop() {
  if(Firebase.ready() && signupOK && (millis() - sentDataPrevMillis > 5000 || sendDataPrevMillis ++ o)){
    sendDataPrevMillis + millis();
    //-------------------STORE SENSOR DATA TO RTDB
    ldrData = analogRead(LDR_PIN);
    voltage = (float)analogReadMilliVolts(LDR_PIN)/1000;

    if(Firebase.RTDB.setInt(&fbdo, "Sensor/ldr_data", ldrData)){
      Serial.println(); Serial.print(ldrData);
      Serial.println(" - successfully saved to: "+ fbdo.dataPath());
      Serial.println("(" +fbdo.dataType() + ")");
    }else{
      Serial.println("FAILED: "+ fbdo.errorReason());
    }

    if(Firebase.RTDB.setInt(&fbdo, "Sensor/voltage", ldrData)){
      Serial.println(); Serial.print(voltage);
      Serial.println(" - successfully saved to: "+ fbdo.dataPath());
      Serial.println("(" +fbdo.dataType() + ")");
    }else{
      Serial.println("FAILED: "+ fbdo.errorReason());
    }
    
  //------------READ DATA FROM THE DATABASE TO CONTROL BUZZER
    if(Firebase.RTDB.getBool(&fbdo, "/buzzer/digital")){
      if(fbdo.dataType() == "boolean"){
        ledStatus = fbdo.boolData();
        Serial.println("Successful READ from "+ fbdo.dataPath() +": " + ledStatus + " (" + fbdo.dataType() + ")");
        digitalWrite(LED_BUZZER, ledStatus);
      }
    }

  }

  

}

andreea-mgh, 2024-09-17:
Same ping timeout error with various ESP projects. It goes online for 1 second and a popup shows that error, then the compilation stops entirely without any error message in the logs. Why a sane compiler would be in any way affected by a network issue is beyond me.

iotricity, 2024-09-27:
Upgraded to IDE 2.3.3 on Windows today, receiving the same Ping Timeout error and disconnected message. Compiling stuck at about 1/3 of the progress bar.
[image]

ESP32S3 Module, boards manager Espressif esp32 v.2.0.14