arduino/arduino-ide #2220: Display output from Arduino Firmware Uploader

per1234, 2023-09-06:
### Describe the request

Print the output from the `arduino-fwuploader` invocations to the IDE's "**Output**" panel.

🙂 It will be easier for the user to troubleshoot problems that might occur during the process.
🙂 Target-specific usage advice will be visible to the user.

### Describe the current behavior

Arduino IDE's ["**Firmware Updater**"](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-fw-cert-uploader#firmware-updater) and ["**Upload SSL Root Certificates**"](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-fw-cert-uploader#certificates-uploader) features allow the user to update the supplemental modules found on some Arduino boards. These features use the [**Arduino Firmware Uploader**](https://arduino.github.io/arduino-fwuploader/) command line tool.

As with any other complex technical operation, the process might fail for a variety of reasons. For example:

- https://github.com/arduino/arduino-ide/issues/1497
- https://github.com/arduino/arduino-ide/issues/750
- https://forum.arduino.cc/t/uno-r4-wifi-board-is-online-but-no-ota-option-for-upload/1152220/33
- https://forum.arduino.cc/t/unable-to-update-firmware-on-arduino-nano-33-iot/1142340

Currently, the only feedback provided to the user when this happens is a generic message with advice that is not always helpful:

> ⚠ Installation failed. Please try again.

Even when the process is successful, it might be that the target-specific plugins print helpful usage advice. For example, this important message printed by Arduino Firmware Updater when targeting the [**UNO R4 WiFi**](https://docs.arduino.cc/hardware/uno-r4-wifi) board:

```text
Upload completed! To complete the update process please disconnect and then reconnect the board.
```

🙁 The user doesn't receive any information about the cause of a failed operation in the Arduino IDE GUI. They must sift through the logs generated by the IDE to find any such information.
🙁 There is no way for important target-specific advice to be displayed to the user in the Arduino IDE GUI.

### Arduino IDE version

73ddbefc3e22c65d00ff59cade57fda14af01818

### Operating system

All

### Operating system version

Any

### Additional context

For the sake of transparency, I suggest also printing the `arduino-fwuploader` command to the output, as is done by Arduino IDE during the compilation and upload processes when verbose output is enabled. This information can be useful for troubleshooting and also helps advanced users to develop workflows using **Arduino Firmware Uploader** directly.

#### Related

- https://github.com/arduino/arduino-ide/issues/2060
- https://github.com/arduino/arduino-ide/issues/786

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

dankeboy36, 2023-09-07:
Great idea!

> Print the output

Do you prefer IDE to print the output at the end of the command, or does it want to provide the terminal feeling? For example, when there is some data on the process's `stdout`/`stderr`, should IDE update the _Output_?

If the latter, there is an overlay when the operation is running, and the dialog is visible. Is IDE happy updating the _Output_ with the overlay, or is there a requirement to change the UX?

per1234, 2023-09-10:
>Do you prefer IDE to print the output at the end of the command, or does it want to provide the terminal feeling? For example, when there is some data on the process's stdout/stderr, should IDE update the Output?

Streaming the Arduino Firmware Uploader output (as is done with the verbose compilation and upload output) would be ideal because it would provide the user with some indication of progress of the operation. But if that makes the feature significantly more difficult to implement then I think it would be fine to print all the collected output after the `arduino-fwuploader` command exits.

>there is an overlay when the operation is running, and the dialog is visible. Is IDE happy updating the Output with the overlay, or is there a requirement to change the UX?

The idea behind this proposal was to leverage the existing user interface of Arduino IDE rather than having to find a way to communicate the information within the "Firmware Update" and "Upload SSL Root Certificates" dialogs.

I understand that it is maybe slightly unintuitive for the output to be printed to the underlying IDE window even though the operation is being performed by the user in a modal dialog. However, I don't think it will be too bad because the user is already accustomed to looking at the Output view for logs from various processes, including Library Manager and Boards Manager.

kittaakos, 2023-09-13:
Here is the feature in action. I see tons of stdout (`%` when flashing), it could be nicer. The stderr variant is more helpful. If IDE2 needs it, I can finalize the feature.

`stdout`:

https://github.com/arduino/arduino-ide/assets/1405703/91a4f29e-93c4-4f7a-8b46-6dfa8104ab79

`stderr`:

https://github.com/arduino/arduino-ide/assets/1405703/6417b4e5-98e1-4597-a669-5926057fffcf

per1234, 2023-09-14:
> I see tons of stdout (% when flashing)

This is due to https://github.com/arduino/arduino-ide/issues/1473

> it could be nicer

Definitely, but I think sub-ideal output is still better than none at all.

>The stderr variant is more helpful.

Just to make sure the requirements are clear, we need both stdout (most importantly because this is where any user instructions such as the need to power cycle the UNO R4 WiFi will be printed) and stderr (because this will be needed for troubleshooting failed attempts).