arduino/arduino-ide #2223: No indication for an unsaved sketch

shobeira, 2023-09-10:
### Describe the problem

In the IDE 2.2.1, there is no indication whether the sketch file is saved already or there is/are unsaved changes that is needed to be saved. The indication was there in IDE v2.1.1

### To reproduce

No need.

### Expected behavior

An indicator next to the sketch file name to show if there is/are unsaved change(s).

### Arduino IDE version

2.2.1

### Operating system

Linux

### Operating system version

22.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-09-10:
Hi @shobeira. Thanks for your report.

> An indicator next to the sketch file name

Are you referring to this indicator?:

[image]

Do you have "**Auto save**" enabled in your Arduino IDE preferences?

Which theme do you have selected from the "**Theme**" menu in your Arduino IDE preferences?

shobeira, 2023-09-11:
Hi @per1234,

I realised it after I submitted the issue. Yea, the auto save is enabled and that explained everything. Maybe a grayed out "autosave" next to the filename would be a good idea though.

closed 2023-09-11 by per1234