arduino/arduino-ide #2224: Keyboard Shortcuts for settings Defaults to PC shortcut when using INTEL Mac, they are correct on M2 an Mac

jeffwrule, 2023-09-11:
### Describe the problem

Settings shortcut (Arduino IDE -> settings...) defaults to non-mac shortcut (ctrl+opt+cmd+P) on INTEL Mac (Macbook Pro 16" 2019 w/ 2.4 GHz 8-Core Intel Core i9.   But the same download installed on an M2 Mac (Mac Book Pro 2023, Apple M2 Max) has the correct CMD+, shortcut for the preferences.   It is minor but annoying.   Both machines are running OS version Ventura 13.5.2 (22G91).   Both Machines are running the same version of Arduino IDE.

### To reproduce

Open the 'Arduino IDE' menu on an Intel Mac; you can see the shortcut from there.  The shortcut is a non-mac shortcut of ctrl+opt+cmd+P

### Expected behavior

should have a cmd+,

### Arduino IDE version

Version: 2.2.1 Date: 2023-08-31T14:26:39.874Z CLI Version: 0.34.0  Copyright © 2023 Arduino SA

### Operating system

macOS

### Operating system version

Ventura 13.5.2 (22G91)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-09-12:
Thanks for reporting!

> Open the 'Arduino IDE' menu on an Intel Mac; you can see the shortcut from there.

I am using a US English input source with a QWERTY keyboard on an Intel Mac and see the expected <kbd>⌘</kbd>+<kbd>,</kbd> with `2.2.1`:

[image]

> non-mac shortcut (ctrl+opt+cmd+P)

Are you saying you have to press four different keys (<kbd>⌃</kbd>+<kbd>⌥</kbd>+<kbd>⌘</kbd>+<kbd>P</kbd>) to open the Settings? If yes, please share a screenshot of how it is shown in the menu on your environment. I want to compare it with the expected behavior.

IDE2 is based on [Eclipse Theia](https://theia-ide.org/). If you could get their blueprint application from https://theia-ide.org/docs/blueprint_download, start it, and see what the shortcut for `File` > `Settings...` > `Settings` is, it would be super helpful. It looks like this on my Intel Mac:

[image]

per1234, 2023-09-28:
Hi @jeffwrule. We are still waiting for you to provide the information requested in the previous reply so that we can proceed with the investigation or evaluation of this.

Please add the requested information in a comment here.