arduino/arduino-ide #2227: arduino ide not starting up after update to 2.21

ekaggrat, 2023-09-16:
### Describe the problem

the arduino ide starts up but if I try to open a new file of simply open a existing file the screen comes up but just the arduino icon blinks in the middle of the screen and the file never opens

### To reproduce

update or install arduino 2.2.1 in windows 10 and open a new file

### Expected behavior

the ide should open a new window with a new file or a existing file

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-09-16:
Hi @ekaggrat. Thanks for your report.

Arduino IDE automatically generates log files while it is running. I'm going to ask you to provide the latest log file from your computer. It might contain information that will help us to identify the problem.

Please do this:

1. Perform the procedure required to put Arduino IDE into the hang state, just as you did before.
1. Click the following link to open the issue page here on GitHub:
   https://github.com/arduino/arduino-ide/issues/2227#issue-comment-box
1. Click the "**Attach files by dragging & dropping, selecting or pasting them**" link at the bottom of the "**Leave a comment**" field.
1. Select the latest `.log` file from this folder (the files are named according to the date):
   ```text
   C:\Users\<username>\AppData\Roaming\Arduino IDE\
   ```
   (where `<username>` is your Windows username)
   ❗ Note that the `AppData` folder is hidden by default. You can make it visible by opening the [Windows "**File Explorer**"](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1) "**View**" menu, then checking the box next to "**☐ Hidden items**".
1. Click the "**Open**" button to attach the file to the comment.
1. You will now see an "**Uploading your files... (1/1)**" status indicator at the bottom of the "**Leave a comment**" field. Wait until that disappears.
1. Click the "**Comment**" button to publish the comment.

---

Please let me know if you have any questions or problems while following those instructions.

AnatolSher, 2023-09-16:
Hello! Version 2.2.1 not starting in win7 64b pro sp1
Entry point in kernel32.dll not found. Log files not created.
[image]
Version 2.1.1 work fine.

per1234, 2023-09-16:
@AnatolSher we do not provide support for using Arduino IDE 2.x with Windows versions older than 10. @ekaggrat is using Windows 10 so your problem is different and off topic for this issue. If you want a solution for your problem, please follow the instructions provided in this forum post:

https://forum.arduino.cc/t/arduino-ide-2-2-0-the-procedure-entry-point-getpackagefamilyname-could-not-be-located-kernel32-dll/1163362/8

If you have any questions or problems while following those, you are welcome to request assistance over on the forum, but any reports specific to Windows 7 are off topic here on GitHub.

AnatolSher, 2023-09-16:
I'm understood, thank you

ekaggrat, 2023-09-17:
i fixed by wiping out arduino completely and deleting all folders and then reinstalling. the issue occurred because I let arduino update from a version 2.0 directly to 2.2.1. 

thanks

kittaakos, 2023-09-21:
Logs from https://github.com/arduino/arduino-ide/issues/2231

[logs-arduino-ide-2231.txt](https://github.com/arduino/arduino-ide/files/12686248/logs-arduino-ide-2231.txt)