arduino/arduino-ide #2230: cannot perform ota update after ota credentials changed

roblatour, 2023-09-18:
### Describe the problem

I have a program for which ota updates were working fine.
device name was A
password was A

In my coded, I then changed these to 
device name B
password B

I did an over the air update, selecting the current device (A) from the port list, and providing the current password (A)

the upload worked fine

now, the program is showing as 
device name B (in the ports list as expected).

However, the Arduino IDE is not prompting me for a password when I try and do a OTA update, rather it is failing with:

Authenticating...FAIL
16:33:39 [ERROR]: Authentication Failed
Failed uploading: uploading error: exit status 1

I tried exiting and restarting the Arduino IDE.

I tried rebooting my computer.

I tried the nightly build of the Arduino IDE.

The same error re-appears.

### To reproduce

As described above

### Expected behavior

If the authentication fails on an OTA update attempt, on the next OTA update attempt the IDE should prompt the user for the password (rather than, I assume, continuing to use a bad cached password).

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

11

### Additional context

this issue seems to be some what similar to https://github.com/arduino/arduino-ide/issues/1443 
However, in this case restarting the IDE does not resolve the issue.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

roblatour, 2023-09-18:
Update: I found a temp manual work around.  

I copied the last line displayed in the Output window that shows the call to the espota.exe program and contains the snippit referenced above.  I changed the word Password to be my new password and ran the command via the Windows cmd prompt.  It worked fine.

However, updates using the Arduino IDE are still failing as noted at the top of this issue under 'Describe the problem'.

roblatour, 2023-09-18:
2nd Update: I found a second, more acceptable, work around.

I move the program folder for my project to a newly created folder.

for example, my old project folder was:  
D:/ArduinoPrograms/myprogram
and my ino file was D:/ArduinoPrograms/myprogram/myprogram.ino 

I moved it to: D:/ArduinoPrograms/myprogram/myprogram
D:/ArduinoPrograms/myprogram/myprogram/myprogram.ino

After which I ran an OTA update, the Arduino IDE prompted me for a password, I entered my new password, and it worked fine.

per1234, 2023-09-21:
>If the authentication fails on an OTA update attempt, on the next OTA update attempt the IDE should prompt the user for the password

This is exactly how the system is designed to work. On the next upload attempt following a failed upload, the "**Configure and Upload**" dialog should appear in Arduino IDE. You are able to update the OTA upload password in that field.

I just did quite a lot of testing of OTA uploads to an ESP32 board using Arduino IDE 2.2.1 and it works just as expected for me every time.

@roblatour please reply here to tell us whether you see a "**Configure and Upload**" dialog appear when you start an upload to a network port after the previous upload attempt failed:

[image]

---

As for workarounds, you can simply select **Sketch > Configure and Upload** from the Arduino IDE menus to force the "**Configure and Upload**" dialog to appear at the start of the upload even if a password is already cached and the previous upload attempt didn't fail.

per1234, 2023-09-21:
I had a breakthrough in the investigation thanks to some information provided in a report of the same problem on the forum (I'm guess that is also you @roblatour???):

https://forum.arduino.cc/t/ide-2-2-1-fails-with-incorrect-ota-password-and-does-not-prompt-for-new-one/1169963

> If a successful OTA upload is performed and then the IDE is closed and later re-opened the IDE appears to "remember" the board/port OTA password combination and attempts to upload the OTA code without asking for an OTA password.

The essential condition that was not mentioned here is that the upload is made to the associated board+port that was automatically selected by Arduino IDE when the sketch is restored on Arduino IDE startup.

### To reproduce

#### Equipment

- Any board that supports upload over a network port.<br />
  Examples:
  - [MKR1000](https://docs.arduino.cc/hardware/mkr-1000-wifi)
  - Any ESP8266-based board
  - Any ESP32-based board
  - [MKR WiFi 1010](https://docs.arduino.cc/hardware/mkr-wifi-1010) (advanced configuration is required)
  - [Nano 33 IoT](https://docs.arduino.cc/hardware/nano-33-iot) (advanced configuration is required)

#### Steps

1. Follow the tutorial to set up your chosen board for uploading over a network port.
   - **MKR1000:** https://docs.arduino.cc/tutorials/mkr-1000-wifi/wifi-101-ota
   - **ESP8266-based board:** - https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#application-example
   - **ESP32-based board:** **File > Examples > ArduinoOTA > BasicOTA**
   - **MKR WiFi 1010** / **Nano 33 IoT:** https://github.com/jandrassy/ArduinoOTA
1. Create and save or open an existing saved sketch.
   **ⓘ** This step is performed in order to associate the board+network port selection with a sketch that can be restored on IDE startup.
1. Wait for the network port to appear in the Arduino IDE ports list.
1. Select the board and network port from the Arduino IDE menus.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start Arduino IDE.
   The saved sketches that were open during the previous session will be restored.
1. Select the window sketch that has the board+network port pre-selected due to the association created in the previous steps.
1. Select **Sketch > Upload** from the Arduino IDE menus.
   🐛 The "**Configure and Upload**" dialog does not appear.
   🐛 If you have verbose upload output enabled, you will see that a default password of "Password" is used in the upload command.
1. Select any other board+port from the Arduino IDE menus.
1. Select the target board+network port in Arduino IDE.
1. Select **Sketch > Upload** from the Arduino IDE menus.
   🙂 The "**Configure and Upload**" dialog does appear.

roblatour, 2023-09-21:
@per1234 I cannot take credit for the post [from ](https://forum.arduino.cc/t/ide-2-2-1-fails-with-incorrect-ota-password-and-does-not-prompt-for-new-one/1169963) by @robgw as that is someone else.  

Regardless, I'm glad you have identified more of what lays at the heart of this, as well as an easier work around (as noted in the post above).

```
That is more complex than is necessary. I'll provide a more simple workaround:

Select any other board from the Tools > Board menu in Arduino IDE.
Select the target board in Arduino IDE once again.
```

Well done.   

Hope a code fix to address this will be possible.

Cheers

closed 2023-09-26 by kittaakos