arduino/arduino-ide #2235: Aeduino IDE 2.2.1 does not want to search (in the boards selector)

The-Young-Maker, 2023-09-21:
### Describe the problem

When I write the name of my board in the Boards windows , It wont show the result or will show nothing .

### To reproduce

I am not sure if i am the only one on which this happened

### Expected behavior

I am not sure if i am the only one on which this happened

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-09-22:
Hi @The-Young-Maker. We will need far more information in order to proceed with the investigation of the problem you are attempting to report. Keep in mind that I can't see your computer and I don't experience any problem resembling what you so vaguely mention here so the only information about the problem I have is what you deem to share here.

Please provide detailed instructions I can follow to reproduce the problem, including:

- What did you do?
- What were the results you expected from doing that thing?
- What were the results you observed that did not match your expectations?

Make sure to include the full and exact text of any error or warning messages you might have encountered.

The-Young-Maker, 2023-09-22:
Hello, yesterday I was in a bit of a hurry but here is some additional information. What I did , was simply open the boards manager (to select my board) and wrote the name of my board.  What I expected was for it to show me the search result, but instead it only shows something like: "    NO BOARDS FOUND FOR ""    " even if the name of my board is correctly written. Also, this problem doesn't always happen, but when it does I have to restart the Arduino IDE so it MAYBE work again.

rboeije1, 2025-05-21:
Just to confirm that in May 2025 and Arduino IDE 2.3.5 this issue is still in the field. I had it some time ago and thought I found a solution (more a workaround) but so far I did not succeed to find it again...