arduino/arduino-ide #2238: IDE 2.2.1 massive CPU usage, slow start on Linux

dronus, 2023-09-24:
### Describe the problem

On Ubuntu 22.04, the IDE version 2.2.1 does start very slowly and continues to use a massive amount of CPU even when idling for quite some while.

- The IDE is the slowest starting application on my system, even with simple platforms (Arduino 8-bit) and small source files to load upon startup.
- While and after starting, it makes my Laptop run loud, hot with massive battery drain. No other application does this (maybe games, but I don't use them). Laptop is a Thinkpad T480s, 4 cores, 8 threads, quite snappy for most applications.
- The massive load sometimes stops after about two minutes after starting, sometimes never. I can't get a clear picture when.
- The process using the most CPU is one of the many processes the IDE launches, the one with the `--renderer` command line option. This loads at minimum a single core for at least 1.5 minutes.
- The processes launched look like this is chromium-based. However, chromium starts and runs fast on my system.
- Starting from terminal, a massive amount of debugging information is printed. The last one about 1 minute 40 seconds after starting, if the IDE is not touched.

### To reproduce

Start the IDE 2.2.1 from the .zip release with a simple project (Arduino 8-bit, eg blink example) opened at the last run. Run from terminal to see output.

### Expected behavior

IDE starting up in a sane amount of time and use no or a miniscule amount of CPU on idling.

### Arduino IDE version

2.2.1

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Additional context

Problems are there since first 2.x release, and never really got better.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-09-25:
Hi @dronus. Thanks for your report.

> Starting from terminal, a massive amount of debugging information is printed. The last one about 1 minute 40 seconds after starting, if the IDE is not touched.

Please provide the text of the output printed to the terminal in a comment here on the GitHub issue. If the quantity exceeds the maximum limit of text for a comment, you can save it to a `.txt` file and add it as an attachment. If you do so, make sure to come here to the issue on the GitHub website (https://github.com/arduino/arduino-ide/issues/2238) as GitHub removes attachments that are sent via email replies.

gicking, 2023-12-26:
Hello,

same issue here (Ubuntu 22.04, Arduino IDE 2.2.1). Stopping the Language Server solved my problem, see [here](https://github.com/arduino/arduino-ide/issues/714#issuecomment-1114452951). Apparently slow startup and heavy CPU load had been an issue in IDE 2.0.0, see [here](https://github.com/arduino/arduino-ide/issues/714). Back then the issue was eventually closed as "solved". So I guess it either was re-introduced again with newer IDE version - or it hadn't been solved for Linux/Ubuntu...?  

For now I can live without Language Server

PaulStoffregen, 2023-12-28:
@gicking - Just curious, can you say how much RAM and what sort of storage device (rotating hard drive, SATA SSD, NVME SSD) your computer has?

I definitely do not see this problem with IDE 2.2.1 running on my Ubuntu 22.04 desktop.  But I'm using 96GB RAM and Intel 905P (the lowest latency NVME on today's market).  Maybe the problem is related to RAM and swapping?  Or maybe it's storage related, requiring more I/O ops per second than higher latency storage can provide?  Or it could be something completely unrelated to hardware.  Just blind guessing, but knowing what sort of hardware you're running could help shine some light on the situation.

gicking, 2023-12-28:
@PaulStoffregen - I have the following setup:

```
CPU: Intel(R) Core(TM) i3-3220 CPU @ 3.30GHz
RAM: 16245MB
SSD: ATA KINGSTON SA400S3
OS: Ubuntu 22.04.3 LTS

```

Attached please find a screenshot of the task manager during indexing. As you can see swapping is not active. 

[image]

Actually I don't believe it's an issue of my hardware, as I didn't have that issue with older versions of the Arduino IDE.

wolly01, 2024-03-13:
I feel like there is definitely an LSP problem somewhere. _Any_ amount of typing induces max fan ramp on my system that persists until typing stops. Otherwise the IDE seems stable and loads up quickly. 

(Gentoo Linux)