arduino/arduino-ide #2247: arduino ide 2.2.1 has issues starting new file

ekaggrat, 2023-10-06:
### Describe the problem

the ide starts up and when i try to open a new file there is only a main page with a hear beat icon.

### To reproduce

start ide and open new file

### Expected behavior

new file should open

### Arduino IDE version

2.2.1/2.2.2 nightly

### Operating system

Windows

### Operating system version

10

### Additional context

this is not a duplicate . the issue has not been resolved .. please don't mark it as duplicate and close the issue.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-10-06:
Hi @ekaggrat.

Please provide a detailed set of instructions I can follow to reproduce the problem. There are several different procedures that could be described as "open new file" so the instructions you provided are not sufficient.

I need to know exactly what I should do at each step. Don't leave anything out no matter how obvious it might seem to you.

per1234, 2023-10-06:
I see now this is a duplicate of https://github.com/arduino/arduino-ide/issues/2227

@ekaggrat this is very disrespectful behavior. Please only submit one report for each bug. Submitting duplicate reports is counterproductive because it means we spend time cleaning up your mess that could have been put towards actually investigating and fixing the bug.

closed 2023-10-06 by per1234