arduino/arduino-ide #2249: Examples from Custom Libraries Renamed to Active Board

ben-qnimble, 2023-10-07:
### Describe the problem

When going to File, Examples, I would expect to see examples broken into a few categories: Built-in examples; Examples from Custom Libraries; and Examples for XXXX where XXXX is the active board selected. I have a fresh install of Arduino IDE 2.2.1 on Windows 11 and all the examples that should be under "Examples from Custom Libraries" and mis-categorized as "Examples for XXXX". If there are board-specific libraries ("Examples for XXXX"), they get merged with the Custom Libraries which can be confusing as it can make it the number of libraries in that category very large and hard to distinguish between the two types of examples.

Here is an example of what I'm seeing:
[image]

And this is an example of what I'd expect:
[image]

Interestingly, I do not see this behavior on a different Windows 11 machine also running Arduino IDE 2.2.1 but, in the setup that works, it was a well-used Arduino IDE install that was upgraded from 2.1.1 (and 2.0.X before that, etc). I'm seeing the issue on a fresh install of Windows 11 and a Arduino IDE 2.2.1 was the first / only Arduino IE install.

### To reproduce

Look at the list of examples (File, Examples) and see if there are are "Examples from Custom Libraries".

### Expected behavior

See above, but I would expect Examples for board and examples from custom libraries to not get merged into one group.

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-10-09:
Thank you for the detailed bug report.

I have read your description multiple times but still need some help understanding the problem. Thanks for your patience.

> all the examples that should be under "Examples from Custom Libraries" and mis-categorized as "Examples for XXXX". If there are board-specific libraries ("Examples for XXXX"), they get merged with the Custom Libraries

I am unaware of any recent changes that will merge example menu items in IDE2.

> ### Expected behavior
> See above, but I would expect Examples for board and examples from custom libraries to not get merged into one group.

I do not understand it; the first screenshot is with an Uno board, and the other is with a MKR WiFi 1010.

There was an issue with the `File` > `Examples` menu group: #2239. Please check whether your problem is the same. If so, the bug has been fixed. Please use the nightly build: https://www.arduino.cc/en/software#nightly-builds

If not, please explain to me once more what's your request. Thanks!

ben-qnimble, 2023-10-09:
I believe this issue is different from #2239 because the example are not missing, they are mis-categoried.  I've also tried the nightly builds with the same behavior. 

The issue shows up with any board as a target, and you can see it when the examples show the generic / common examples labelled as "Examples for XXX" instead of "Examples from Custom Libraries". But to make it more clear in the worst case if your target board hardware has examples, it combines them with these generics. Below is a screenshot showing this behavior which combines these examples and the proper behavior which separates generic examples for all board and from board-specific examples.

[image]

[image]

I hope that explains what's going on. Please let me know if you need more details.

kittaakos, 2023-10-09:
Thanks for the screenshots; now I see what's the problem. I could not reproduce it with 2.2.1 on macOS. Please help me. 

> The issue shows up with any board as a target,

Do you mean sometimes the example menu groups are correct, and sometimes they aren't? I am asking because you have a screenshot from 2.2.1 which is correct–I assume. Can you provide me with the precise steps for what you do when you see the correct menu groups and what you don't?

ben-qnimble, 2023-10-09:
So I have two windows 11 machines. One is brand new and I installed Arduino IDE fresh on it and it shows the problem always. The other (upgraded over the years from maybe earlier Arduino IDE versions never has the problem. It is easy to see the program because when you select any target board, the second set of examples are labelled "Examples for XXXX" where XXXX is the selected target board. What it should show (and shows on my other machine) is "Examples from Custom Libraries". When the target board does not have any examples, then the only difference (that I see) is the label for the second set of examples (the first being 'build-in examples'). However, when the target is a board that does have examples is selected, then the difference is the merging of these examples into one category instead of separating "Examples from Custom Libraries" and "Examples for XXXX". 

As for how to reproduce, I don't fully know. I wonder if the behavior has to do with a preference / configuration issue that maybe got set on my working machine with an old version of Arduino that persisted when upgrading to 2.21. That said, since I see the problem on a fresh install of 2.2.1 on a fresh install of Windows (and no customization / configuration other than adding a board manager url), I think it is a problem going forward. Maybe installing in a clean VirtualBox would enable it to be reproduced. 

If you can think of things I can try to narrow down when / why this issue shows up, please let me know.

kittaakos, 2023-10-10:
Thanks for your persistence.

> So I have two windows 11 machines.

Please do the following on both machines:
 - Open a `CMD.EXE` and navigate to your IDE2 installations. On my machine, when I am in the IDE2 installation folder, I see this when I execute `dir`:
```
c:\Users\kittaakos\Desktop\ide2 2.2.1>dir
 Volume in drive C has no label.
 Volume Serial Number is 04F3-13CC

 Directory of c:\Users\kittaakos\Desktop\ide2 2.2.1

10/10/2023  08:54 AM    <DIR>          .
10/10/2023  08:54 AM    <DIR>          ..
08/31/2023  02:36 PM       162,060,992 Arduino IDE.exe
08/31/2023  02:35 PM           129,690 chrome_100_percent.pak
08/31/2023  02:35 PM           179,971 chrome_200_percent.pak
08/31/2023  02:35 PM         4,891,080 d3dcompiler_47.dll
08/31/2023  02:35 PM         2,862,080 ffmpeg.dll
08/31/2023  02:35 PM        10,541,296 icudtl.dat
08/31/2023  02:35 PM           479,232 libEGL.dll
08/31/2023  02:35 PM         7,514,112 libGLESv2.dll
08/31/2023  02:35 PM             1,096 LICENSE.electron.txt
08/31/2023  02:35 PM         6,762,963 LICENSES.chromium.html
10/10/2023  08:54 AM    <DIR>          locales
10/10/2023  08:54 AM    <DIR>          resources
08/31/2023  02:35 PM         5,430,329 resources.pak
08/31/2023  02:35 PM           162,352 snapshot_blob.bin
08/31/2023  02:35 PM           476,792 v8_context_snapshot.bin
08/31/2023  02:35 PM         5,209,088 vk_swiftshader.dll
08/31/2023  02:35 PM               106 vk_swiftshader_icd.json
08/31/2023  02:35 PM           920,576 vulkan-1.dll
              16 File(s)    207,621,755 bytes
               4 Dir(s)  180,013,531,136 bytes free
```
 - Run the following command and copy the complete terminal output either here or, if it's too long, put it into a file and attach it to this issue.
```
resources\\app\\lib\\backend\\resources\\arduino-cli.exe lib list --all --fqbn arduino:samd:mkrwifi1010 --format json
```

<details><summary>Click to see the output from my Windows machine</summary>
<p>

```json
[
  {
    "library": {
      "name": "LiquidCrystal",
      "author": "Arduino, Adafruit",
      "maintainer": "Arduino \u003cinfo@arduino.cc\u003e",
      "sentence": "Allows communication with alphanumerical liquid crystal displays (LCDs).",
      "paragraph": "This library allows an Arduino/Genuino board to control LiquidCrystal displays (LCDs) based on the Hitachi HD44780 (or a compatible) chipset, which is found on most text-based LCDs. The library works with in either 4 or 8 bit mode (i.e. using 4 or 8 data lines in addition to the rs, enable, and, optionally, the rw control lines).",
      "website": "http://www.arduino.cc/en/Reference/LiquidCrystal",
      "category": "Display",
      "architectures": [
        "*"
      ],
      "install_dir": "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal",
      "source_dir": "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal\\src",
      "version": "1.0.7",
      "license": "Unspecified",
      "location": 1,
      "layout": 1,
      "examples": [
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal\\examples\\Autoscroll",
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal\\examples\\Blink",
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal\\examples\\Cursor",
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal\\examples\\CustomCharacter",
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\LiquidCrystal\\examples\\Display",
[... 874 lines cut ...]
      "version": "1.0.2",
      "license": "Unspecified",
      "location": 1,
      "layout": 1,
      "examples": [
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\WiFi101OTA\\examples\\WiFi101_OTA",
        "C:\\Users\\kittaakos\\Documents\\Arduino\\libraries\\WiFi101OTA\\examples\\WiFi101_SD_OTA"
      ],
      "provides_includes": [
        "InternalStorage.h",
        "OTAStorage.h",
        "SDStorage.h",
        "WiFi101OTA.h"
      ],
      "compatible_with": {
        "arduino:samd:mkrwifi1010": true
      }
    }
  }
]
```
</p>
</details>

The above command will list all installed libraries for the Arduino MKR WiFi 1010 board. IDE2 does very similar thing under the hood to build the `File` > `Examples` menu structure.

ben-qnimble, 2023-10-10:
On the machine where it isn't working correctly, when I run that, I get
```
{
  "error": "Error listing libraries: Unknown FQBN: platform arduino:samd is not installed"
}
```
If I select arduino:avr:uno instead, which is installed on my setup, I get

<Details>
<Summary>On machine with issue</Summary>

```
{
    "library": {
      "name": "SPI",
      "author": "Arduino",
      "maintainer": "Arduino \u003cinfo@arduino.cc\u003e",
      "sentence": "Enables the communication with devices that use the Serial Peripheral Interface (SPI) Bus.",
      "paragraph": "SPI is a synchronous serial data protocol used by microcontrollers for communicating with one or more peripheral devices quickly over short distances. It uses three lines common to all devices (MISO, MOSI and SCK) and one specific for each device.",
      "website": "http://www.arduino.cc/en/Reference/SPI",
      "category": "Communication",
      "architectures": [
        "avr"
      ],
      "install_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SPI",
      "source_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SPI\\src",
      "container_platform": "arduino:avr@1.8.6",
      "version": "1.0",
      "license": "Unspecified",
      "location": 2,
      "layout": 1,
      "examples": [
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SPI\\examples\\BarometricPressureSensor",
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SPI\\examples\\DigitalPotControl"
      ],
      "provides_includes": [
        "SPI.h"
      ],
      "compatible_with": {
        "arduino:avr:uno": true
      }
    }
  },
  {
    "library": {
      "name": "SoftwareSerial",
      "author": "Arduino",
      "maintainer": "Arduino \u003cinfo@arduino.cc\u003e",
      "sentence": "Enables serial communication on any digital pin.",
      "paragraph": "The SoftwareSerial library has been developed to allow serial communication on any digital pin of the board, using software to replicate the functionality of the hardware UART. It is possible to have multiple software serial ports with speeds up to 115200 bps.",
      "website": "http://www.arduino.cc/en/Reference/SoftwareSerial",
      "category": "Communication",
      "architectures": [
        "avr"
      ],
      "install_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SoftwareSerial",
      "source_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SoftwareSerial\\src",
      "container_platform": "arduino:avr@1.8.6",
      "version": "1.0",
      "license": "Unspecified",
      "location": 2,
      "layout": 1,
      "examples": [
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SoftwareSerial\\examples\\SoftwareSerialExample",
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\SoftwareSerial\\examples\\TwoPortReceive"
      ],
      "provides_includes": [
        "SoftwareSerial.h"
      ],
      "compatible_with": {
[... 523 lines cut ...]
      "version": "0.1",
      "license": "Unspecified",
      "location": 2,
      "examples": [
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks\\examples\\OnDemandADCTrigger",
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks\\examples\\UseExternalADCClock",
        "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks\\examples\\UseExternalClock"
      ],
      "provides_includes": [
        "main.h"
      ],
      "compatible_with": {
        "qnimble:imxrt:quarto": true
      }
    }
  }
]

```

</details>

I'll post the same results on my other machine (without the issue) in a second comment. 

For the record, this output list looks right to me. There's no "LiquidCrystal" or "Ethernet" examples that show up under "Examples for Quarto" in the IDE.

kittaakos, 2023-10-11:
> Machine with issue on board with examples (Quarto)

> Quarto (working right>>

I created the diff between the two responses after sorting the object via `.library.name`:

<details><summary>Click to see the diff</summary>
<p>

```diff
12,14c12,14
<       "install_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\analog_output",
<       "source_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\analog_output",
<       "container_platform": "qnimble:imxrt@1.8.4",
---
>       "install_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\analog_output",
>       "source_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\analog_output",
>       "container_platform": "qnimble:imxrt@1.9.0",
19,22c19,22
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\analog_output\\examples\\DualChannelDDS",
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\analog_output\\examples\\DualChannelDDS_Dynamic",
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\analog_output\\examples\\OutputPatternOnTrigger",
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\analog_output\\examples\\PreCalculated_DualChannelDDS"
---
>         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\analog_output\\examples\\DualChannelDDS",
>         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\analog_output\\examples\\DualChannelDDS_Dynamic",
>         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\analog_output\\examples\\OutputPatternOnTrigger",
>         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\analog_output\\examples\\PreCalculated_DualChannelDDS"
33a34,59
>       "name": "Arduino_BuiltIn",
>       "author": "Arduino",
>       "maintainer": "Arduino <info@arduino.cc>",
>       "sentence": "Shell Arduino library to provide the standard built-in libraries",
>       "paragraph": "The sole purpose of this library is to install the standard Arduino libraries which are typically included with the Arduino IDE.",
>       "website": "https://github.com/arduino-libraries/Arduino_BuiltIn",
>       "category": "Other",
>       "architectures": [
>         "*"
>       ],
>       "install_dir": "C:\\Users\\bluey\\Documents\\Arduino\\libraries\\Arduino_BuiltIn",
>       "source_dir": "C:\\Users\\bluey\\Documents\\Arduino\\libraries\\Arduino_BuiltIn\\src",
>       "version": "1.0.0",
>       "license": "Unspecified",
>       "location": 1,
>       "layout": 1,
>       "provides_includes": [
>         "Arduino_BuiltIn.h"
>       ],
>       "compatible_with": {
>         "qnimble:imxrt:quarto": true
>       }
>     }
>   },
>   {
>     "library": {
42,44c68,70
<       "install_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks",
<       "source_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks",
<       "container_platform": "qnimble:imxrt@1.8.4",
---
>       "install_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\clocks",
>       "source_dir": "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\clocks",
>       "container_platform": "qnimble:imxrt@1.9.0",
49,51c75,77
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks\\examples\\OnDemandADCTrigger",
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks\\examples\\UseExternalADCClock",
<         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.8.4\\libraries\\clocks\\examples\\UseExternalClock"
---
>         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\clocks\\examples\\OnDemandADCTrigger",
>         "C:\\Users\\bluey\\AppData\\Local\\Arduino15\\packages\\qnimble\\hardware\\imxrt\\1.9.0\\libraries\\clocks\\examples\\UseExternalADCClock",
[... 453 lines cut ...]
>         "macros.h",
>         "masstorage.h",
>         "max3421e.h",
>         "max_LCD.h",
>         "message.h",
>         "parsetools.h",
>         "printhex.h",
>         "settings.h",
>         "sink_parser.h",
>         "usb_ch9.h",
>         "usbh_midi.h",
>         "usbhid.h",
>         "usbhost.h",
>         "usbhub.h",
>         "version_helper.h",
>         "xboxEnums.h"
>       ],
>       "compatible_with": {
>         "qnimble:imxrt:quarto": true
>       }
>     }
```

</p>
</details>

ben-qnimble, 2023-10-11:
Sorry, I had different version of the board package on the two machines from some testing. But when I change to the same version, that part of the diff goes away, but the same behavior remains.

Looking at just which libraries are listed from the `arduino-cli.exe lib list` command, I found some differences. When running with a FQBN of arduino:avr:uno, on the machine with the issues, I see far few libraries installed compared to the working version. Here's the diff in the list of libraries between the two machines (extra libraries on the machine that works as expected):

``` diff
+Arduino_BuiltIn
 EEPROM
+Ethernet
 HID
+Keyboard
+LiquidCrystal
+Mouse
 SPI
 SoftwareSerial
+Stepper
+TFT
+USB Host Shield Library 2.0
 Wire
```

Similarly, on my other FQBN that has its own examples, I get the same behavior with more libraries listed on the machine that is working properly: 

``` diff
 Analog Output
+Arduino_BuiltIn
 Clocks
 Demos
+Ethernet
+LiquidCrystal
 Lock In
+Mouse
 RTClib_min
 SD
 SPI
 SdFat
 Servo
+Stepper
+TFT
 Testing
 Time
+USB Host Shield Library 2.0
 qCommand
 qProtothreads
```

It seems like on the problem machine some Arduino Builtin libraries are not listed from the lib list. Why this causes those same libraries to later get added by the IDE and combined into the 'Examples from XXXX' board, I don't know, but is sure seems like a clue. On the machine that isn't acting right, I've only installed the Arduino AVR Boards (1.8.6) and my board support package for the Quarto. (Although removing Quarto and only having the AVR Boards installed has the same problem / behavior).

For reference, here's the python code I wrote to sort the json list by library name and to output just the library names. The above diffs are diffs of those library lists.
<details>
<summary>Python Script</summary>

``` python
import json

fn = 'BadQuarto.txt'

def sort_json(filename):
    f = open(filename)
    data = json.load(f)

    new_data = sorted(data,key = lambda d: d["library"]["name"])

    newfile = fn[:-4] + '-Sorted.txt'
    newlibs = fn[:-4] + '-Libs.txt'

    with open(newfile, 'w') as fout:
        json.dump(new_data,fout,indent=4)
   
    with open(newlibs, 'w') as fout:
        for d in new_data:
            fout.write('{}\n'.format(d['library']['name']))

sort_json(fn)
```
</details>

per1234, 2023-10-13:
## Overview of Examples Menu Categorization

There are three different categories of locations where Arduino libraries are installed:

- **Platform** (e.g., `C:\Users\bluey\AppData\Local\Arduino15\packages\qnimble\hardware\imxrt\1.9.0\libraries\`)
- **Built-in** (e.g., `C:\Users\bluey\AppData\Local\Arduino15\libraries\`)
  **ⓘ** Despite the similar name, this is completely different from the [standalone example sketches](https://github.com/arduino/arduino-examples) that are listed under the "**Built-in examples**" section of the menu.
- **Sketchbook** (e.g., `C:\Users\bluey\Documents\Arduino\libraries\`)

The libraries under the platform location are typically bundled with the platform due to being very specific to the architecture of the boards of that platform. Only the libraries of the specific platform of the currently selected board are accessible at any given time.

The libraries under the built-in location are the libraries that are automatically installed by Arduino IDE on the first run after a fresh installation. These are fundamental official libraries which are installed by default in order to provide new users with a gentle learning curve by allowing them to make a decent range of projects before introducing the requirement to learn how to use Library Manager.

The libraries under the sketchbook folder are the ones the user has installed or updated (via **Library Manager**, "[**Add .ZIP Library...**](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#importing-a-zip-library)", or [manual installation](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#manual-installation)).

The example sketches bundled with the installed libraries are listed in the **File > Examples** menu. This menu is separated into sections. The section under which a given library is placed in the menu depends on its location category:

|Location|Menu Section|
|-|-|
|platform|"**Examples for \<board name\>**"|
|built-in|"**Examples for \<board name\>**"|
|sketchbook|"**Examples from Custom Libraries**"|

## Explanation of Reported Behaviro

Let's focus on a single library that is in a location under the **File > Examples** menu unexpected to you. I think that once you understand the reason for the location of that library, you will also understand the situation with all the other libraries.

In the screenshots you shared [here](https://github.com/arduino/arduino-ide/issues/2249#issuecomment-1753078454), the "Ethernet" library examples are located under the "**Examples for Quarto**" section of the menu in the first screenshot, and under the "**Examples from Custom Libraries**" section in the second screenshot.

In the `arduino-cli lib list` output under the "Quarto (working right>>" section of [this comment](https://github.com/arduino/arduino-ide/issues/2249#issuecomment-1755491337), we see the `install_dir` value for the "Ethernet" library is:

```json
"C:\\Users\\bluey\\Documents\\Arduino\\libraries\\Ethernet"`
```

Confusingly, we don't see an entry for the "Ethernet" library in the `arduino-cli lib list` output under the "Machine with issue on board with examples (Quarto)" section of [this comment](https://github.com/arduino/arduino-ide/issues/2249#issuecomment-1755465594). The reason for this is that the `arduino-cli lib list` command only returns a list of libraries installed in the **platform** and **sketchbook** location categories. The only way to get a list of the libraries in the **built-in** location is using [the gRPC interface of Arduino CLI](https://arduino.github.io/arduino-cli/dev/rpc/commands/#librarylistrequest), as Arduino IDE does. If the data for the libraries in the "built-in" location was included in the `arduino-cli lib list` output, you would see that the `install_dir` value for the "Ethernet" library on that machine is:

```json
"C:\\Users\\bluey\\AppData\\Local\\Arduino15\\libraries\\Ethernet"
```

So the reason why the "Ethernet" library is listed in a different section of the **Examples** menu on one of your machines compared to another is that the library is installed at a different location on each machine. The reason for that is likely that you installed "Ethernet" via the Arduino IDE Library Manager on the "Machine with issue" machine at some point (which might even have been before you installed Arduino IDE 2.x if you were using 1.x previously), whereas on the "working right" machine, the only installation of the library is the one installed automatically on the first run via the IDE's "built-in" library installation system.

## Conclusion

Since it seems everything is working as intended, I will close this issue.

I do think that the IDE's current behavior of using the "**Examples for \<board name\>**" section of the menu for libraries from the "built-in" location is not ideal. Since this issue is quite convoluted and rambling, I don't think it can effectively serve to track the proposal of changing that behavior so I created a new issue for that purpose: https://github.com/arduino/arduino-ide/issues/2259

## Additional Concepts

Note that a library of a given name might be installed to multiple locations. In this case, [Arduino CLI determines which of those libraries is the best match for the currently selected board](https://arduino.github.io/arduino-cli/dev/sketch-build-process/#dependency-resolution).

### Library Manager

[... cut ...]
Only the platform bundled libraries from the platform of the currently selected board are available. In this case, [Arduino CLI determines which of those libraries is the best match for the currently selected board](https://arduino.github.io/arduino-cli/dev/sketch-build-process/#dependency-resolution). This means that the menu item for a given library name might actually be originating from a completely different library installation, which might have a different location category, depending on which board you have selected.

So you might see a given library name move from one section of the menu to another after you select a different board from the Arduino IDE menus.

closed 2023-10-13 by per1234

Dmitry-Chernikov, 2023-12-05:
I don't have any libraries of the Mega 2560 platform at all, I don't have an EEPROM library, although there is one in the arduino15 package.
[image]