arduino/arduino-ide #2254: Uploading with multiple programmers attached does not work

CreativeRobotics, 2023-10-10:
### Describe the problem

I have a setup with two boards, each with an embedded debugger (Curiosity Nano dev boards with ATTiny 3227 MCU and the Mega Tiny Core)
Both are recognised as comm ports and I can work with two Arduino instances using both serial interfaces.
If I try and upload with "Upload using programmer" it will always use the same programmer regardless of which Arduino instance I am using.

### To reproduce

Connect two curiosity nano boards
Launch two instances of Arduino
Upload using programmer from one instance
Upload from the second instance

### Expected behavior

This ought to behave the same way that it does when using two boards and app instances with USB or serial bootloaders.
Arduino should be able to recognise if multiple programmers are attached as USB devices, just as it recognises multiple COMM ports.
Multiple debugger/programmers should be selectable for each Arduino instance, as happens with COMM ports.

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

10

### Additional context

I am developing a board to board comm system so it helps to have both boards connected. The devices are too small for bootloaders so I have to use the programmers. If they had serial bootloaders then this setup would work properly because each instance would be connected to a different serial port.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-10-11:
Hi @CreativeRobotics. If you enable **Show verbose output during: `☑` compilation** in the Arduino IDE "**Preferences**" dialog and then attempt an upload to your Curiosity Nano board, you will see that the `avrdude` command doesn't even use the serial port. For example:

```text
"C:\Users\per\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude" "-CC:\Users\per\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/avrdude.conf" -v -V -pattiny817 -ccuriosity_updi -Pusb "-Ufuse0:w:0b00000000:m" "-Ufuse2:w:0x02:m" "-Ufuse5:w:0b11000101:m" "-Ufuse6:w:0x04:m" "-Ufuse7:w:0x00:m" "-Ufuse8:w:0x00:m" "-Uflash:w:C:\Users\per\AppData\Local\Temp\arduino\sketches\42A552BDC21CE9BF12E4B456461B61AE/27.ino.hex:i"
```

This is because nEDBG programmer that is used for the upload is a pure USB device, which is identified in the upload command only by using its VID/PID pair (`03eb:2175`) and AVRDUDE does not support identifying it using a serial port number (you will see the `-P` port flag is instead set to `usb` in the command. The problem is that every nEDBG has the same VID/PID pair, so this upload command doesn't give the [**AVRDUDE** uploader tool](https://avrdudes.github.io/avrdude/) any way to know which specific board to target when multiple devices exist on your system with that VID/PID pair.

There is sometimes a mechanism for uniquely identifying specific hardware instances. In the case of serial ports, this is a serial number. AVRDUDE supports the use of the serial number in the upload command via `-Pusb:<serial number>` flag:

https://avrdudes.github.io/avrdude/7.2/avrdude_3.html#Option-Descriptions:~:text=be%20specified%20as-,usb%5B%3Aserialno%5D,-.%20In%20that%20case

Modern versions of Arduino IDE do make the serial number associated with a selected serial port accessible for use in the upload command via the `upload.port.properties.serialNumber` property:

https://arduino.github.io/arduino-cli/latest/platform-specification/#pluggable-discovery

So in theory this change to the "**megaTinyCore**" platform could solve the problem of the `avrdude` command being ambiguous in regards to targeted board when multiple are connected to the computer:

```patch
diff --git a/megaavr/platform.txt b/megaavr/platform.txt
index bbf02bf7..c4de0e83 100644
--- a/megaavr/platform.txt
+++ b/megaavr/platform.txt
@@ -174,6 +174,15 @@ tools.avrdude.program.params.quiet=-q -q
 tools.avrdude.program.params.noverify=-V
 tools.avrdude.program.pattern={upload.prog_interlock}"{cmd.path}" "-C{config.path}" {program.verbose} {program.verify} -p{build.mcu} -c{protocol} {program.extra_params} "-Ufuse0:w:{bootloader.WDTCFG}:m" "-Ufuse2:w:{bootloader.OSCCFG}:m" "-Ufuse5:w:{bootloader.SYSCFG0}:m" "-Ufuse6:w:{bootloader.SYSCFG1}:m" "-Ufuse7:w:{bootloader.APPEND}:m" "-Ufuse8:w:{bootloader.BOOTEND}:m" "-Uflash:w:{build.path}/{build.project_name}.hex:i"
 
+## This tool targets the specific device instance with the serial number obtained via the selected serial port.
+tools.avrdude-with-sn.path={runtime.tools.avrdude.path}
+tools.avrdude-with-sn.cmd.path={path}/bin/avrdude
+tools.avrdude-with-sn.config.path={runtime.platform.path}/avrdude.conf
+tools.avrdude-with-sn.program.params.verbose=-v
+tools.avrdude-with-sn.program.params.quiet=-q -q
+tools.avrdude-with-sn.program.params.noverify=-V
+tools.avrdude-with-sn.program.pattern={upload.prog_interlock}"{cmd.path}" "-C{config.path}" {program.verbose} {program.verify} -p{build.mcu} -c{protocol} -Pusb:{upload.port.properties.serialNumber} "-Ufuse0:w:{bootloader.WDTCFG}:m" "-Ufuse2:w:{bootloader.OSCCFG}:m" "-Ufuse5:w:{bootloader.SYSCFG0}:m" "-Ufuse6:w:{bootloader.SYSCFG1}:m" "-Ufuse7:w:{bootloader.APPEND}:m" "-Ufuse8:w:{bootloader.BOOTEND}:m" "-Uflash:w:{build.path}/{build.project_name}.hex:i"
+
 ## Used for "upload" through bootloader
 tools.avrdude.upload.params.verbose=-v
 tools.avrdude.upload.params.quiet=-q -q
diff --git a/megaavr/programmers.txt b/megaavr/programmers.txt
index cdc7aa20..ba497a59 100644
--- a/megaavr/programmers.txt
+++ b/megaavr/programmers.txt
@@ -59,7 +59,10 @@ nedbg.name=Curiosity Nano (nEDBG, debug chip: ATSAMD21E18)
 nedbg.communication=usb
 nedbg.protocol=curiosity_updi
 nedbg.program.protocol=curiosity_updi
+# This tool recipe will be used as a fallback by development tools that don't have pluggable discovery support.
 nedbg.program.tool=avrdude
+# This tool recipe will be used by development tools that do have pluggable discovery support.
+nedbg.program.tool.default=avrdude-with-sn
 nedbg.program.extra_params=-Pusb
 
 edbg.name=Xplained Pro (EDBG, debug chip: AT32UC3A4256)
```

I have used this successfully with other devices but I don't own an nEDBG device so I can't test it. I do own an [ATtiny817 Xplained Mini](https://www.microchip.com/en-us/development-tool/attiny817-xmini), which has the mEDBG instead of the nEDBG. I tried applying the patch above to the "**Xplained Mini (mEDBG, debug chip: ATmega32u4)**" programmer of **megaTinyCore**. Unfortunately I find that the upload fails:

```text
"C:\Users\per\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude" "-CC:\Users\per\AppData\Local\Arduino15\packages\megaTinyCore\hardware\megaavr\2.6.10/avrdude.conf" -v -V -pattiny817 -cxplainedmini_updi -Pusb:ATML2658061800005922 "-Ufuse0:w:0b00000000:m" "-Ufuse2:w:0x02:m" "-Ufuse5:w:0b11000101:m" "-Ufuse6:w:0x04:m" "-Ufuse7:w:0x00:m" "-Ufuse8:w:0x00:m" "-Uflash:w:C:\Users\per\AppData\Local\T [... cut ...]
I don't know whether that same problem occurs with the nEDBG.

What I do know is that this has nothing to do with the Arduino IDE codebase hosted in this repository so I will close this as off topic. If you have any questions or would like to discuss the subject further, feel free to make a topic on Arduino Forum:

https://forum.arduino.cc/

---

Semi-related (but different since in this case we do have a discovered port that provides us with the serial number): https://github.com/arduino/arduino-cli/issues/770

closed 2023-10-11 by per1234

CreativeRobotics, 2023-10-11:
Thanks, I suspected this might be the case, but I thought it was worth posting an issue just in case there was a fix.