arduino/arduino-ide #2263: shortcut keys broken on non-US/qwerty layouts

viorels, 2023-10-23:
On Romanian keyboard layout ctrl-F triggers "code format" (as ctrl-T) instead of "find".
This layout is very similar to US layout, except some AltGr combinations, so this should not happen.

Steps to reproduce:
1. Set keyboard layout to Romanian/Romanian
2. Open Arduino
3. Open a sketch and hit ctrl-f
Expected "find" dialog, but results in code format

Similar shortcut keys being switched happen on US/Colemak layout (ctrl-S and ctrl-R are mixed). But maybe I should report that in a different bug?

Ubuntu 22.04, Arduino IDE 2.2.1

tobiasBora, 2024-12-10:
Can confirm this is extremely annoying. In the french layout, A is Q, so typing Ctrl-A to select everything will… close the program without warning (dataloss here!). Similarly, copy/pasting in bépo produce unwanted behaviors.

Dev6050, 2025-07-05:
I’m seeing the same issue on macOS.
When a non‑English keyboard layout is active and you open the Arduino IDE, the IDE’s keyboard shortcuts get remapped incorrectly.
For example, the Copy shortcut—normally ``⌘`` + ``C``—is changed to ``⌃`` + ``⌥`` + ``⌘`` + ``C`` under a non‑English layout.