arduino/arduino-ide #2266: Serial monitor copy not functional on MacOS

frankcohen, 2023-10-26:
### Describe the problem

Create a sketch, compile, upload, view Serial Monitor, click anywhere in the Serial Monitor panel, get no response. I am hunting for a way to copy text. For example, I compile, upload, and run a new sketch and see this in the Serial Monitor panel:

```
Info, Reflections of Frank
Starting new log file, forcecount = 0
Opened log file /log333, mylog.size() = 41
Info, Host: REFLECTIONS-88
Info, mjpegFile = /a130f0c903a0472cd043cd2252a5f3a4/a130f0c903a0472cd043cd2252a5f3a4.mjpeg
Guru Meditation Error: Core  1 panic’ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420e6c31  PS      : 0x00060b30  A0      : 0x820079b9  A1      : 0x3fcebf20  
A2      : 0x3fca2770  A3      : 0xb5ac8f7d  A4      : 0xb5ac8f7d  A5      : 0x3fca51e0  
A6      : 0x3fca2e70  A7      : 0xff000000  A8      : 0x82007896  A9      : 0x3fcebed0  
A10     : 0x00000000  A11     : 0x0011e37f  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3fca51e0  A15     : 0x00000001  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x420e6c2e:0x3fcebf20 0x420079b6:0x3fcebf40 0x42007e5f:0x3fcebf60 0x42047f52:0x3fcebfd0
```
I roll the mouse over the above Backtrace text. Click and drag or right double click. I get no activity. No text selection.

### To reproduce

Create a sketch
Enter source code
Compile and upload
View Serial Monitor
Click and drag within the service monitor.
No selection made.

### Expected behavior

Highlighted selected text appears within the Serial Monitor panel.

### Arduino IDE version

arduino-ide_nightly-20231025_macOS_64bit

### Operating system

macOS

### Operating system version

Sonoma 14.0 (23A344)

### Additional context

Same behavior in:
arduino-ide_nightly-20230804_macOS_64bit
and Arduino IDE 2.2.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-10-26:
Thanks for your report @frankcohen. I tried it on my macOS Sonoma machine and found that highlighting by doing a click and drag in the Serial Monitor view works as expected:

[image]

Please try to replicate the click and drag action shown in my screencast above and then comment here to tell us whether the Serial Monitor content is highlighted.

There are some known bugs related to highlighting text in Serial Monitor:

- https://github.com/arduino/arduino-ide/issues/247
- https://github.com/arduino/arduino-ide/issues/1081

However, neither of those seem to me to match the behavior you describe.

frankcohen, 2023-10-26:
Thank you for the reply and video. Here is what I am seeing... 

https://github.com/arduino/arduino-ide/assets/6834993/92d92a40-bcf6-4870-9261-21fe2aba57e1

I get the same behavior using Night Builds: October 25, 2023 and August 15, 2023.

I don't think it's a hardware issue, yet I should tell you this is on my open-source Reflections dev board, using ESP32-S3 and JTAG over USB
https://github.com/frankcohen/ReflectionsOS/blob/main/Docs/images/CindyLou.jpg

Here are the board settings in Arduino IDE 2.2.1:
https://github.com/frankcohen/ReflectionsOS/blob/main/Docs/images/ESP32-S3_Board_Settings.jpg

I'm glad to help in any way.

-Frank

per1234, 2023-10-26:
Thanks! That screencast was very valuable. I see now that this is a bug in the "Light High Contrast" theme you are using:

[image]

Notice that when any of the other themes are selected, the highlight is visible.

The low level functionality of the highlight works fine even though it is not visible. You can copy the invisibly highlighted text to the clipboard as usual.

frankcohen, 2023-10-27:
Awesome, I confirm that switching to Light theme and I am able to select regions of text in the Serial Monitor panel normally. I tried it on the Dark and Dark High Contract themes and they work correctly too. Just my luck! lol -Frank

P.S. - I just got my appointment for cataract surgery this morning. Thank the team for the high contrast themes, they help me a lot.