arduino/arduino-ide #2268: Arduino says sketch should be in subdirectory called ProffieOS, which it already is.

profezzorn, 2023-10-27:
### Describe the problem

I have a file called "ProffieOS.ino". It is in a directory called "ProffieOS".
I run the arduno 2.2.1 app image and regardless of how I try to open this ino file, it complains that it needs to be in a subdirectory called "ProffieOS", then offers to move the file, which would mess up all my include paths.

It seems unlikely that this would happen to everybody all the time, but I haven't found an easy way to figure out *why* Arduino thinks that my ProffieOS folder isn't named ProffieOS.

I think there must be a file inside my ProffieOS directory which is confusing Arduino, because when I download a fresh copy of ProffieOS Arduino seems to open it just fine. However, if that is the case, then Arduino should just say so.

### To reproduce

arduino-ide_2.2.1_LInux_64bit.AppImage ProffieOS/ProffieOS.ino

### Expected behavior

Arduino should recognize that the sketch is already in a directory with the right name.

### Arduino IDE version

2.2.1

### Operating system

Linux

### Operating system version

Ubuntu 20.05.5 LTS

### Additional context

The full path name for my ProffieOS folder is /net/cytocin/home/hubbe/hack/teensy/lightsaber/ProffieOS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-10-27:
Hi @profezzorn. Thanks for your report.

I note the `/net/` in the path. Does the fault still occur if you move the bad copy of the sketch to a location that is not under `/net/`? When you downloaded the fresh copy of the sketch that did not produce the fault, was its path under `/net/`?

profezzorn, 2023-10-27:
/net/ is in fact a normal directory in this case, it's not a magic NFS mount or anything.
All my files are under /net/, including the directory where I downloaded, unzipped and tested the plain copy of ProffieOS.

profezzorn, 2023-10-27:
These are the files/folders under the ProffieOS directory with the problem:

blades/
borked-color
buttons/
common/
config/
display/
doc/
elf.cc~
fontconvert/
functions/
ir/
libs/
LICENCE.txt
Makefile
motion/
mtp/
pov_tools/
pqoi/
ProffieOS.ino
props/
README.md
scripts/
show_notworking
show_working
sound/
styles/
transitions/
v6_announce.txt
v6_announce.txt~
v7.x-diff
videotoblc/
zipfiles/

None of these seems to have "meaningful" names to me.
I suppose arduino might be looking in subdirectories for something?

profezzorn, 2023-10-27:
The problem seems to be a file called ProffieOS/blades/.#pwm_pin.h
When I remove that file, Arduino is able to open the sketch.
This file is created by emacs while editing a file, and can sometimes be left behind.
Not sure why Arduino thinks that "ProffieOS" isn't called "ProffieOS" because it contains a .# file though...

VeloSteve, 2023-11-25:
I have this problem in a much simpler situation.  The code is here if you want to see it:
https://github.com/VeloSteve/CBASS-Arduino-Sketch

A simple sketch for setting a real time clock works fine on my desktop, and the version on GitHub has no recent changes.  Today I cloned the repository and compiled with a 1.8 version.  No problem.  Then I installed 2.2.1 and immediately started getting this message.

I need to use 2.2.1, if only because when students download the IDE they will get that version by default, and it needs to work for them.

The only thing odd here is that the files are not in the default sketch directory from the install, but this hasn't been a problem before.  The path is

D:\GitHub\CBASS-Arduino-Sketch\CBASS_Clocksetter

The only things in the directory are the two *.ini files and a README.md.

per1234, 2023-11-26:
Hi @VeloSteve. The problem you experienced has a different cause than the problem reported by **@

profezzorn, 2023-12-13:
This is happening again.
I don't have any ".#*" files this time.
Tried a nightly build, which also didn't help.

I'm going to go make a backup, then delete one file at a time until it starts working again.....

profezzorn, 2023-12-13:
Never mind, I do have a .#something file again, I just missed it the first three times I looked.

econeale, 2024-01-08:
I have a similar problem with IDE 2.2.1 on Windows. In my case, I had to remove a sub-folder containing test sketches. It looks like the IDE is trying to recursively open sketches and complains if it finds extra INO files. Have you looked at your scripts directory for such things?

andreh7, 2024-07-21:
FWIW, On MacOS I see a similar problem when I had invalid softlinks in the project directory (the IDE 2.3.2 wanted to move `proj/proj.ino` to `proj/proj/proj.ino`.

To reproduce I can do:

```
ln -s a b
```

in `proj/` and open `proj/proj.ino` with the Arduino IDE.

```
/Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources/arduino-cli" upload proj.ino
```

showed an error message like:

```
b: no such file or directory
```

so I guess some part of the code scans the directory and fails when a softlink points to 'nowwhere' and leads to a confusing error message.