arduino/arduino-ide #2269: Error message not displayed when Boards Manager installation fails with "UNKNOWN" status code

per1234, 2023-10-28:
### Describe the problem

When a Boards Manager installation fails due to a platform or tool archive file download failing with an "UNKNOWN" [status code](https://grpc.github.io/grpc/core/md_doc_statuscodes.html), Arduino IDE does not display the error message provided by Arduino CLI.

🐛 The lack of this important information will make it difficult to troubleshoot and support the failure.

### To reproduce

1. Create a file named `package_foo_index.json` with the following content:
   ```json
   {
     "packages": [
       {
         "name": "foo",
         "maintainer": "nobody",
         "websiteURL": "https://github.com/arduino/arduino-ide/issues/2269",
         "email": "nobody@example.com",
         "platforms": [
           {
             "name": "arduino/arduino-ide#2269 demo",
             "architecture": "bar",
             "version": "0.0.0",
             "category": "Contributed",
             "help": {
               "online": "http://example.com/"
             },
             "url": "https://expired.badssl.com/",
             "archiveFileName": "baz",
             "checksum": "SHA-256:203d2717c7004209e38a498f35761c23f47f4447640ba130f5fe48f753ddd581",
             "size": "123",
             "boards": [],
             "toolsDependencies": []
           }
         ],
         "tools": []
       }
     ]
   }
   ```
1. Select **File > Preferences...** (or **Arduino IDE > Settings...** for macOS users) from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Enter a URL with [`file` scheme](https://en.wikipedia.org/wiki/File_URI_scheme) pointing to the path of the `package_foo_index.json` on your computer into the "**Additional Boards Manager URLs**" field in the "**Preferences**" dialog.
   For example, if the file was at `e:\stuff\package_foo_index.json`, you would enter the following URL:
   ```text
   file:///e:/stuff/package_foo_index.json
   ```
   ❗ If there are already Boards Manager URLs in the field, separate them with commas.
1. Click the "**OK**" button.
1. You will now see a "**Downloading index: ...**" notification at the bottom right corner of the IDE window. Wait for that notification to close.
1. Select **Tools > Board > Boards Manager** from the Arduino IDE menus to open the "**Boards Manager**" view in the left side panel.
1. Scroll down through the list of boards platforms until you see the "**arduino/arduino-ide#2269 demo**" entry.
1. Click the "**INSTALL**" button at the bottom of the entry.

🐛 Arduino IDE does not display any information about the installation failure:

Notification:

> **Failed to install platform: 'arduino/arduino-ide#2269 demo:0.0.0'. 2 UNKNOWN:**

Output view:

```text
Downloading packages
foo:bar@0.0.0
Failed to install platform: 'foo:bar:0.0.0'.
Error: 2 UNKNOWN: 
```

### Expected behavior

Arduino IDE displays useful error message content when a failure occurs.

In the demo above, the expected behavior is to print the error message:

```text
Get "https://expired.badssl.com/": tls: failed to verify certificate: x509: certificate has expired or is not yet valid:
```

### Arduino IDE version

69b7365

### Operating system

Windows

### Operating system version

11

### Additional context

I bisected the regression to https://github.com/arduino/arduino-ide/pull/2199 (fault does not occur when using the build from the previous commit, 7f660d7).

---

When the `arduino.cli.daemon.debug` [setting](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings) is enabled, you can see from the logs that Arduino IDE does receive the message from Arduino CLI:

```text
2023-10-28 05:47:03 2023-10-28T12:47:03.762Z daemon INFO 59 CALLED: /cc.arduino.cli.commands.v1.ArduinoCoreService/PlatformInstall STREAM_RESP
2023-10-28 05:47:03 2023-10-28T12:47:03.763Z daemon INFO 59 |  REQ:  {
59 |    "instance": {
59 |      "id": 1
59 |    },
59 |    "platform_package": "foo",
59 |    "architecture": "bar",
59 |    "version": "0.0.0"
59 |  }
2023-10-28 05:47:03 2023-10-28T12:47:03.764Z daemon INFO 59 |  RESP: {
59 |    "task_progress": {
59 |      "name": "Downloading packages"
59 |    }
59 |  }
time="2023-10-28T05:47:03-07:00" level=info msg="Starting download" url="https://expired.badssl.com/"
59 |  RESP: {
59 |    "progress": {
59 |      "Message": {
59 |        "Start": {
59 |          "url": "https://expired.badssl.com/",
59 |          "label": "foo:bar@0.0.0"
59 |        }
59 |      }
59 |    }
59 |  }
2023-10-28 05:47:03 2023-10-28T12:47:03.978Z daemon INFO 59 |  RESP: {
59 |    "progress": {
59 |      "Message": {
59 |        "End": {
59 |          "message": "Get \"https://expired.badssl.com/\": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: "
59 |        }
59 |      }
59 |    }
59 |  }
59 |  ERROR:  Get "https://expired.badssl.com/": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: 
59 STREAM CLOSED
2023-10-28 05:47:03 2023-10-28T12:47:03.979Z discovery-log INFO start
2023-10-28 05:47:03 2023-10-28T12:47:03.979Z discovery-log INFO start new deferred
2023-10-28 05:47:03 2023-10-28T12:47:03.979Z root ERROR Request install failed with error: 2 UNKNOWN:  Error: 2 UNKNOWN: 
    at t.callErrorFromStatus (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:724760)
    at Object.onReceiveStatus (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:752377)
    at Object.onReceiveStatus (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:745401)
    at C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:831523
    at process.processTicksAndRejections (node:internal/process/task_queues:77:11)
for call at
    at a.makeServerStreamRequest (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:752144)
    at a.platformInstall (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:816420)
    at b.install (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:84350)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async u.onRequest (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:1068332)
    at async c.handleRequest (C:\arduino-tools\231-69b7365\resources\app\lib\backend\main.js:2:1059227)
```

---

I used the relatively rare conditions of downloading a platform archive file from a server with an expired SSL certificate for the demo since that was how I happened to encounter the fault, and is easy to reproduce. However, I suspect that the fault might also occur under other conditions where the information from the error message will be even more important.

There are regular reports on Arduino Forum of mysterious failures that have an "UNKNOWN" status code:

https://forum.arduino.cc/search?expanded=true&q=%22Error%3A%202%20UNKNOWN%22

These would be even more difficult to support without this information.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

giacomocusinato, 2024-02-23:
Tested both on MacOS 14.3 
- Locally at https://github.com/arduino/arduino-ide/commit/aa9b10d68e6653b7f40ab920173c7964aaefbfa4 
- Release version `2.3.2`

The error shows correctly as a notification and in the output view. Perhaps might be a Windows problem?

per1234, 2024-02-24:
> Perhaps might be a Windows problem?

I confirm.

I am still able to reproduce the fault using the build from https://github.com/arduino/arduino-ide/commit/aa9b10d68e6653b7f40ab920173c7964aaefbfa4 on two separate Windows machines by following the instructions from the issue report, but I have now checked on my macOS and Linux machines and the error message is shown as expected.