arduino/arduino-ide #2278: conditional expressions and debugging

kandoooo, 2023-11-07:
### Describe the problem

Using an Arduino Zero
Plug in and ensure board configured with IDE 2.2.1
Load a simple example - Digital -Button.
compile and start debug from the debug icon just below tools menu
at line 42 or within the loop somewhere add int x=0;
somewhere below that Add if you can a conditional breakpoint with an expression lets say x=1
run the debug and see where it stops it probably won't
now see if you can edit the condition
now see if it stops anywhere.
There seems to be a lot of bugs in the conditional breaks or watch conditions

### To reproduce

Too many

### Expected behavior

I expect first of all that a variable being selected can be used as a debug watched variable without having to type it in.
I expect the conditional break to right click on the left of the line number to actually give me something to type into.
I expect the same for watches.
I expect that if I haven't typed it in correctly that the IDE will tell me that something is wrong straight away.
I expect some examples of what you can type into a conditional box to be available.
I don't expect to have to reload a sketch just to get a type-in box
I really don't expect to have three debug icons to choose from and if  I choose the wrong one I get nothing or another gdb-server tab appearing.
These are just a few of my expectations having paid for an Arduino Zero for the specific reason of debugging.

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

11

### Additional context

I hope the IDE will come out soon with a new version with a lot of these bugs fixed.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-11-08:
> Using an Arduino Zero
> Plug in and ensure board configured with IDE 2.2.1
> Load a simple example - Digital -Button.
> compile and start debug from the debug icon just below tools menu

> somewhere below that Add if you can a conditional breakpoint with an expression lets say x=1
> run the debug and see where it stops it probably won't

It works for me with the Arduino Zero with IDE 2.2.1. Did you select `Sketch` > `Optimize for Debugging` before the upload? See here: https://docs.arduino.cc/tutorials/zero/debugging-with-zero.

> ### To reproduce
> Too many

I could not reproduce it with 2.2.1.

> ### Expected behavior
> I expect first of all that a variable being selected can be used as a debug watched variable without having to type it in. I expect the conditional break to right click on the [...]

Thanks for taking the time and summing all these up. Not all are bugs or IDE2 imperfections; I will focus solely on your conditional breakpoints defect.

If you have not selected `Optimize for Debugging`, please do so and see if the conditional breakpoints work. If you have chosen it, and hitting the breakpoints does not work, please tell me the procedure you do to start the debugging. You could share your sketch or attach a screencast to this issue to understand the nature of the problem better.

kandoooo, 2023-11-10:
For instance: -
1. ensure Zero board plugged in and correct port etc
2. I load the blink example
3. ensure that Optimize for debugging has a green tick next to it
4. click on verify/compile
5. click on upload which compiles and uploads to zero
6. click debug from just below tools menu which shows the debug panel
7. program is now running. I select two breakpoints setup and loop lines and the program halts. "Paused on breakpoints" shows in the threads panel.
8. right click to place a conditional breakpoint or breakpoint from the drop down menu on any other line and nothing happens.

Thank you for not thinking that I'm just time waster and for not getting upset with my comments.
I  hope this helps. Kando

kandoooo, 2023-11-12:
You can also try this:
Use the StateChangeDetection.ino  IDE example
Remove the remarks as Line numbers given are after Top line remarks removed.
Function breakpoint from the breakpoint panel + using the word: - loop
On line 50 place Conditional breakpoint  Hit Count from drop down menu =25
On line 41 place Conditional breakpoint  using these words: - buttonState=LOW

I can't get these 3 conditions in.

kittaakos, 2023-11-23:
See (the gif) how VS Code supports conditional breakpoints from the _Breakpoints_ view: https://code.visualstudio.com/docs/editor/debugging#_conditional-breakpoints

Theia needs catch up with the new debugger UX