arduino/arduino-ide #2279: Poor contrast between highlighted text and selection background in Suggest menu

per1234, 2023-11-09:
### Describe the problem

Arduino IDE has [a "Suggest" (AKA autocomplete/Intellisense) feature](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-autocomplete-feature#the-autocomplete-feature). When triggered, a menu is displayed offering suggested completions for the line at the cursor position in the editor.

This menu uses several colors to convey information, including:

- The part of the suggestion text that has already been written in the editor is indicated by a special text foreground color
- The item in the suggestion list menu currently selected is indicated by a special background color

🐛 When the "Light" theme is in use, the light blue color used by the former has very poor contrast against the light teal color used by the latter.

### To reproduce

1. Select **File > Preferences** from the Arduino IDE menus.
1. Select "**Light**" from the "**Theme**" menu in the "**Preferences**" dialog.
1. Click the "**OK**" button.
1. Select **Tools > Board > Arduino UNO** from the Arduino IDE menus.
1. Type `delay` on a line in the editor.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Space**</kbd> keyboard shortcut ("**Trigger Suggest**).

🐛 The `delay` text in the selected suggestion menu item is very difficult to read due to poor contrast:

[image]

### Expected behavior

All foreground colors used by officially supported themes have good contrast against all possible background colors for that element.

### Arduino IDE version

69b7365

### Operating system

- Windows
- macOS

### Operating system version

- Windows 11
- macOS Sonoma

### Additional context

I bisected the regression to 9a6a457 / https://github.com/arduino/arduino-ide/pull/2144 (the fault does not occur when using the build from the previous commit e17472e)

---

I noticed that the fault does not occur on the first run after clearing the "user data" folder:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Rename or delete (:warning: cautiously) the "user data" folder, which is located at the following path:
   - **Windows**:
     ```
     C:\Users\<username>\AppData\Roaming\arduino-ide
     ```
     (where `<username>` is your Windows username)
   - **Linux**:
     ```
     /home/<username>/.config/arduino-ide/
     ```
     (where `<username>` is your Linux username)
   - **macOS**:
     ```
     /Users/<username>/Library/Application Support/arduino-ide/
     ```
     (where `<username>` is your macOS username)
1. Start Arduino IDE.
1. Perform the steps described in the "**To reproduce**" section of this report.
   🙂 The colors have good contrast, as was the case when using versions of Arduino IDE prior to the regression.
1. Select **File > Quit** from the Arduino IDE menus
1. Start Arduino IDE.
1. Perform the steps described in the "**To reproduce**" section of this report.
   🐛 The colors now have poor contrast.

---

Originally reported by @ubidefeo

---

<a name="workaround"></a>

#### Workaround

Select a different theme in the Arduino IDE preferences.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2023-11-09:
I'd like to add that this styling issue also occurs when using the Command Palette (CMD+SHIFT+P) or Jump to Symbol (CMD+SHIFT+O)

[image]
[image]

daxliniere, 2025-02-09:
Thanks @per1234!
Any idea why it's taking over a year to make this trivial change that everybody sees on a daily basis?