arduino/arduino-ide #2286: After permission denied error on serial port, still says "uploading" and "done uploading"

php4fan, 2023-11-19:
### Describe the problem

I have connected a Sparkfun ESP8266 Thing Dev board, which apparently is connected to `/dev/ttyUSB0`.

The documentation says you need to add yourself to the `dialout` group (and then log out and log in). I tried that, but I got an error saying the `dialout` group doesn't exist. In other forums I read you need to belong to the group `tty`, so I tried that and logged out and in.

Still, I get the permission denied error when uploading a sketch. But this report is not about the error per se, it's about the fact that I get the permission denied error in the console, **and yet**, after that, I also get a progress bar on the bottom right that says "Uploading" and lasts for several seconds, and after that, a balloon that says "Done uploading", with an "i" icon and no error sign, which looks exactly like it has successfully finished uploading. And the sketch definitely hasn't been uploaded.

### To reproduce

Upload a sketch 

### Expected behavior

Either there is an error writing to the port or there isn't.

If there is an error (e.g. Permission Denied), then I should see the error message, and I should _not_ see a message that says "Done uploading". 

If I see a message at the very end that says "Done uploading", then it should mean that it has succesfully uploaded the sketch.

### Arduino IDE version

2.2.1

### Operating system

Linux

### Operating system version

Manjaro Linux

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Nick-Norton, 2023-11-21:
Does the permission group for dialout exist for the Manjaro Distro (awesome of course).
For an Arch based distro, maybe sudo usermod -a -G uucp $USER might work?
I'm no IT professional, just a dumb user.

php4fan, 2023-11-21:
> For an Arch based distro, maybe sudo usermod -a -G uucp $USER might work?

Indeed, that worked and should be added to the documentation.
However, as I said, this report is about the inconsistency between the permission error being shown and the popup still saying "uploading" and "upload done"