arduino/arduino-ide #2304: Nightly Build is completely broken. Empty window on startup

php4fan, 2023-12-09:
### Describe the problem

I was using Arduino 2.2.1 and it mostly works.

Because of [this](https://github.com/dankeboy36/esp-exception-decoder/issues/10#issuecomment-1729496765), I downloaded the nightly build.

When I launch the nightly build, it starts, it loads the last sketch that I was working on, for a fraction of a second you can see it, and then the whole window of the Arduino IDE goes completely empty and doesn't respond.

[image]

### To reproduce

Launch Arduino

### Expected behavior

Should work

### Arduino IDE version

2.2.2

### Operating system

Linux

### Operating system version

Manjaro Linux

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-12-12:
Hi @php4fan. I'm going to ask you to post some additional information that might help us to identify the problem.

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Open [a command line terminal](https://ubuntu.com/tutorials/command-line-for-beginners) in the folder that contains the Arduino IDE installation.
1. Invoke the Arduino IDE executable.
   For example, if you are using the "AppImage" package, the invocation would look like this:
   ```text
   ./arduino-ide_<version>_Linux_64bit.AppImage
   ```
   (where `<version>` is the Arduino IDE version number)
   Or if you are using the ZIP package, the invocation would look like this:
   ```text
   ./arduino-ide
   ```
   Debug output should now be printed to the PowerShell window as Arduino IDE starts (or fails to start).
1. Wait for Arduino IDE to finish starting, or failing to start.
1. Switch back to the terminal window if the Arduino IDE window took the focus.
1. Use the mouse to select all the text in the terminal window. Make sure to scroll all the way up to the top.
1. Press <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**C**</kbd> to copy the selected text to the clipboard.
1. Paste (<kbd>**Ctrl**</kbd>+<kbd>**V**</kbd>) the copied output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

per1234, 2025-01-15:
Closing due to lack of response.

@php4fan if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2025-01-15 by per1234