arduino/arduino-ide #2306: Arduino-IDE will not get past pulsating loading screen on macOS Sonoma 14.1.1 (Apple Silicon)

yayoi-k, 2023-12-11:
### Describe the problem

Stuck at the pulsating loading screen.  

[image]

### To reproduce

Install Arduino IDE 2.2.1 arm64 and launch it.  Or try nightly nightly-20231210 which yields the same results.

### Expected behavior

Application should launch and allow loading a sketch or selecting a device

### Arduino IDE version

2.2.1

### Operating system

macOS

### Operating system version

Sonoma 14.1.1

### Additional context

Arduino-IDE has never worked, I installed it for the first time this evening and got these results.

Here is a paste of the end of the log file:

    at /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/main.js:2:585956
    at E.evaluateErrorStatus (/Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/main.js:2:585972)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async s.<anonymous> (/Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/main.js:2:585642)
2023-12-10 19:33:57 2023-12-11T03:33:57.685Z root INFO Deploy plugins list: 79.3 ms [Finished 2.633 s after backend start]
2023-12-10 19:33:59 Opening channel for service path '/services/electron-window'.
2023-12-10 19:34:00 Opening channel for service path '/services/ide-updater'.
2023-12-10 19:34:01 2023-12-11T03:34:01.220Z root WARN Frontend a.configure took longer than the expected maximum 100 milliseconds: 639.1 ms [Finished 4.541 s after frontend start]
2023-12-10 19:34:01 2023-12-11T03:34:01.292Z root WARN A command editor.action.toggleStickyScroll is already registered.
2023-12-10 19:34:01 2023-12-11T03:34:01.360Z root WARN Frontend keybindings.onStart took longer than the expected maximum 100 milliseconds: 108.2 ms [Finished 4.728 s after frontend start]
2023-12-10 19:37:36 Closing channel on service path '/services/electron-window'.
2023-12-10 19:37:36 Closing channel on service path '/services/ide-updater'.
2023-12-10 19:37:36 Stored workspaces roots: 
2023-12-10 19:37:36 No sketches were scheduled for deletion.
2023-12-10 19:39:24 Using browser-only version of superagent in non-browser environment
2023-12-10 19:39:24 Configuration directory URI: 'file:///Users/chris/.arduinoIDE'
2023-12-10 19:39:24 Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-12-10 19:39:24 2023-12-11T03:39:24.295Z root INFO Backend Object.initialize: 2.4 ms [Finished 0.302 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.295Z root INFO Backend Object.configure: 1.9 ms [Finished 0.302 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.295Z root INFO Backend a.onStart: 0.7 ms [Finished 0.302 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.295Z root INFO Backend a.configure: 0.2 ms [Finished 0.318 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.296Z root INFO Backend u.onStart: 0.2 ms [Finished 0.318 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.296Z root INFO Backend d.configure: 0.1 ms [Finished 0.318 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.297Z root INFO Backend d.initialize: 15.8 ms [Finished 0.318 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.297Z root INFO Backend c.onStart: 0.2 ms [Finished 0.318 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.297Z root INFO Backend s.configure: 0.5 ms [Finished 0.319 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z config INFO >>> Initializing CLI configuration...
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend c.initialize: 0.2 ms [Finished 0.319 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend P.onStart: 0.4 ms [Finished 0.319 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z config INFO Loading CLI configuration from /Users/chris/.arduinoIDE/arduino-cli.yaml...
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/resources/arduino-cli...
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend s.initialize: 0.2 ms [Finished 0.319 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend w.onStart: 0.5 ms [Finished 0.320 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend r.configure: 0.9 ms [Finished 0.320 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z discovery-log INFO start
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z discovery-log INFO start new deferred
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend d.initialize: 0.2 ms [Finished 0.320 s after backend start]
2023-12-10 19:39:24 2023-12-11T03:39:24.298Z root INFO Backend y.onStart: 0.9 ms [Finished 0.321 s after backend start]
[... 204 lines cut ...]
[platform-index] - code: 9, message: Loading index file: loading json index file /Users/chris/Library/Arduino15/package_index.json: open /Users/chris/Library/Arduino15/package_index.json: no such file or directory
[platform-index] - code: 9, message: Error loading hardware platform: discovery builtin:serial-discovery not found
[platform-index] - code: 9, message: Error loading hardware platform: discovery builtin:mdns-discovery not found
[library-index] - code: 9, message: Loading index file: reading library_index.json: open /Users/chris/Library/Arduino15/library_index.json: no such file or directory

2023-12-10 19:45:08 2023-12-11T03:45:08.131Z root ERROR Error: The index of the cores and libraries must be updated before initializing the core gRPC client.
The following problems were detected during the gRPC client initialization:
[platform-index] - code: 9, message: Loading index file: loading json index file /Users/chris/Library/Arduino15/package_index.json: open /Users/chris/Library/Arduino15/package_index.json: no such file or directory
[platform-index] - code: 9, message: Error loading hardware platform: discovery builtin:serial-discovery not found
[platform-index] - code: 9, message: Error loading hardware platform: discovery builtin:mdns-discovery not found
[library-index] - code: 9, message: Loading index file: reading library_index.json: open /Users/chris/Library/Arduino15/library_index.json: no such file or directory

    at /Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/main.js:2:586141
    at _.evaluateErrorStatus (/Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/main.js:2:586157)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async s.<anonymous> (/Applications/Arduino IDE.app/Contents/Resources/app/lib/backend/main.js:2:585827)
2023-12-10 19:45:08 2023-12-11T03:45:08.600Z root INFO Deploy plugins list: 28.2 ms [Finished 1.179 s after backend start]
2023-12-10 19:45:08 2023-12-11T03:45:08.897Z root WARN Frontend a.configure took longer than the expected maximum 100 milliseconds: 198.5 ms [Finished 0.993 s after frontend start]
2023-12-10 19:45:08 2023-12-11T03:45:08.899Z root WARN A command editor.action.toggleStickyScroll is already registered.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-12-11:
Hi @yayoi-k. On the first run after a fresh installation, Arduino IDE downloads some additional essential components from the Internet and installs them. If it can't install those components then the startup fails. So the computer must have access to the Internet on the first startup.

The errors indicate Arduino IDE is not able to access the Internet. If it is not already, please connect your computer to the Internet, close the Arduino IDE window, then start Arduino IDE again. This time it should start up successfully.

After that first run, an Internet connection is not required. However, there are some supplemental features such as "**Library Manager**" that have an inherent dependency on the Internet. Those features will not be usable while your computer is offline. But the essential functionality of editing, compiling, and uploading sketches will be available even without an Internet connection.

---

Although a dependency on an Internet connection is expected, the IDE should handle the lack of a connection gracefully and clearly communicate about the problem to the user. So the current behavior of hanging on startup under these conditions is considered a bug. We are tracking that bug at https://github.com/arduino/arduino-ide/issues/2042, and the associated work to improve the communication about this requirement at https://github.com/arduino/arduino-ide/issues/784. So I'll close this issue as a duplicate of the others.

closed 2023-12-11 by per1234

yayoi-k, 2023-12-11:
Thanks for your response.  I defiantly have internet the entire time, that's also how I downloaded the IDE.  I think there is some bug where IDE can't access the internet via the computer it's running on, or does not have 'permissions' in some way (such as new Apple security change, etc)

yayoi-k, 2023-12-11:
@per1234 please re-open this issue.  I have perfectly working internet the entire time, so the two bugs related to communicating check/failure of internet is not relevant to my issue and would only be providing additional information to the user.  My issue is Arduino IDE unable to use internet, or falsely believes it has no internet.

per1234, 2023-12-11:
@yayoi-k please post over on the Arduino Forum. That will be a more suitable place to support this type of problem:

https://forum.arduino.cc/c/software/ide-2-x/93

I'll try to provide assistance over there. If we determine the problem is actually caused by an untracked bug in Arduino IDE, I'll be happy to reopen this issue but at this time it seems likely this is either a duplicate of one of the existing issues or perhaps a misconfiguration of your system.