arduino/arduino-ide #2313: Arduino 2.2.1 Failed to start the backend application,  do not show any window (UI) at Macos m2

raiscui, 2023-12-21:
### Describe the problem

I has arduino 1.8 long time , today i install 2.2.1, and then  , remove 1.8 ,  then , open the arduino2.2.1 , I found that only the top menu was displayed, but no window was created

MacBook Air
Apple M2
14.1.2 (23B92) Sonoma
[2023-12-21_log.log](https://github.com/arduino/arduino-ide/files/13736679/2023-12-21_log.log)

### To reproduce

I just delete from apps, reinstall 2.2.1  many times , In the end, it's all the same

### Expected behavior

show a Arduino GUI window at screen.

### Arduino IDE version

2.2.1

### Operating system

macOS

### Operating system version

14.1.2 (23B92) Sonoma

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2024-01-15:
Thanks for reporting it.

> 2023-12-21_log.log

It's great that you have attached the logs. I looked into it and could not find anything useful in it due to a bug:

> 2023-12-21 13:30:51 2023-12-21T05:30:51.512Z root ERROR Failed to start the backend application:

The good news is that it has been fixed (https://github.com/arduino/arduino-ide/pull/2236). If you want to help us, please get the nightly IDE build with the logging fixes from here: https://www.arduino.cc/en/software#nightly-builds.

Start the IDE (_it won't start_), get the logs, and attach them again to this issue. Then, we will see what the problem is. 

The nightly build does not have the fix for your startup issue, but we will hopefully see the cause of the issue.

Thanks

raiscui, 2024-01-29:
Thank you for your reply. I ran nightly buildings and did provide me with more information,

Here's the problem:
```
2024-01-29 11:25:26 2024-01-29T03:25:26.356Z config INFO Validating the CLI configuration...
2024-01-29 11:25:26 2024-01-29T03:25:26.415Z config INFO The CLI config is valid.
2024-01-29 11:25:26 2024-01-29T03:25:26.415Z config INFO <<< Initialized the CLI configuration.
2024-01-29 11:25:26 2024-01-29T03:25:26.474Z root INFO Resolve plugins list: 245.2 ms [Finished 1.612 s after backend start]
2024-01-29 11:25:27 2024-01-29T03:25:27.078Z daemon INFO time="2024-01-29T11:25:26+08:00" level=info msg="Using config file: /Users/cuiluming/.arduinoIDE/arduino-cli.yaml"
time="2024-01-29T11:25:26+08:00" level=info msg="arduino-cli version 0.35.1"
time="2024-01-29T11:25:26+08:00" level=info msg="Executing `arduino-cli daemon`"
Daemon is now listening on 127.0.0.1:61168
{"IP":"127.0.0.1","Port":"61168"}
2024-01-29 11:25:27 2024-01-29T03:25:27.078Z daemon INFO Daemon is running.
2024-01-29 11:25:27 2024-01-29T03:25:27.808Z root INFO Deploy plugins list: 245.2 ms [Finished 2.948 s after backend start]
2024-01-29 11:25:31 2024-01-29T03:25:31.080Z root ERROR Failed to start the backend application:
2024-01-29 11:25:31 2024-01-29T03:25:31.080Z root ERROR Error: getaddrinfo ENOTFOUND localhost
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:108:26)
2024-01-29 11:25:31 2024-01-29T03:25:31.083Z root ERROR Uncaught Exception:  Error: getaddrinfo ENOTFOUND localhost
2024-01-29 11:25:31 2024-01-29T03:25:31.083Z root ERROR Error: getaddrinfo ENOTFOUND localhost
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:108:26)
2024-01-29 11:30:54 Stored workspaces roots: 
2024-01-29 11:30:54 No sketches were scheduled for deletion.
2024-01-29 11:30:54 Could not terminate the backend process. It was not running.

```
I don't know what caused it, but I successfully launched the GUI by adding line '127.0.0.1 localhost' to the system hosts.