arduino/arduino-ide #2314: IntelliSense not working on Cloud Project

Tynopia, 2023-12-24:
### Describe the problem

When working with the Arduino IDE on a Cloud Project, IntelliSense functionality is not functioning properly. While it works seamlessly in the local environment when a board is selected, the same is not true for Cloud Sketches. In the latter case, the Code Completion feature either performs poorly or fails to load altogether, irrespective of whether a board is selected or not.

As soon as the code has been in the "Loading..." status for a long time, I get this error and only get bad code completion like in the picture below

*Permanent loading*

[image]

*Failing*

[image]

*Bad Completion*

[image]

### To reproduce

1. Open the Arduino IDE.
2. Create or open a Cloud Project.
3. Attempt to use Code Completion features.
4. Observe the Loading... state or inadequate Code Completion.

### Expected behavior

When working on a Cloud Project in the Arduino IDE, the Code Completion should function smoothly, providing relevant suggestions and not getting stuck in a perpetual loading state. The behavior should be consistent with the local environment, where selecting a board ensures proper Code Completion.

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

21H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Tynopia, 2023-12-26:
It seems like there also might be an issue with the information reference or the ALT navigation.

[image]

kittaakos, 2024-01-15:
Thanks for reporting it. I tested it with 2.2.1, and IntelliSense works fine with cloud sketches (on macOS).

This is a generic issue with the language server, not the cloud sketches—for example, https://github.com/arduino/arduino-ide/issues/1042#issuecomment-1577935224 or [Windows 11 Error: write EOF: undefined](https://forum.arduino.cc/t/windows-11-error-write-eof-undefined/1171291).