arduino/arduino-ide #2318: ARDUINO_USER_AGENT environment variable not set

PaulStoffregen, 2023-12-28:
### Describe the problem

Latest IDE nightly builds are not setting the ARDUINO_USER_AGENT environment variable when running tools.

### To reproduce

Using Linux x86-64:

1: File - Preferences, add Teensy URL 
2: Boards Manager, Install Teensy 0.59.4
3: Copy [this modified teensy_size tool](https://www.pjrc.com/tmp/arduino_user_agent_missing/teensy_size) to ~/.arduino15/packages/teensy/tools/teensy-tools/0.59.4/
4: Click Verify and look for "ARDUINO_USER_AGENT = " in the output panel.

[image]

Alternately, edit the source code for any tool to print the ARDUINO_USER_AGENT environment variable.  This copy of teensy_size merely has these lines added:

``` c
      const char *ardua = getenv("ARDUINO_USER_AGENT");
      fprintf(stderr, "ARDUINO_USER_AGENT = %s\n", ardua);
```

### Expected behavior

The ARDUINO_USER_AGENT environment variable is supposed to be set, according to the Platform Specification, under "Environment variables":

https://arduino.github.io/arduino-cli/0.35/platform-specification/#environment-variables

Arduino IDE 2.2.1 does it correctly:

[image]

### Arduino IDE version

arduino-ide_nightly-20231227_Linux_64bit

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2024-01-15:
Thank you for reporting it

> Latest IDE nightly builds are not setting the ARDUINO_USER_AGENT environment variable

IDE2 has never set any environment variables. I see this is an issue in the CLI: https://github.com/arduino/arduino-cli/issues/2499. I keep this issue open for visibility.

PaulStoffregen, 2024-01-16:
I tested the new CLI 0.35.1 release just now.  Can confirm it defines ARDUINO_USER_AGENT.

Will test IDE nightly builds to confirm here.  Hopefully nightly builds have CLI 0.35.1 soon?

kittaakos, 2024-01-17:
> 3: Copy [this modified teensy_size tool](https://www.pjrc.com/tmp/arduino_user_agent_missing/teensy_size) to ~/.arduino15/packages/teensy/tools/teensy-tools/0.59.4/

@PaulStoffregen, can you provide a link to the modified tool for macOS (Intel)? I want to verify #2332.

 - I downloaded the tool,
 - copied it to the desired location (on macOS `/Users/a.kitta/Library/Arduino15/packages/teensy/tools/teensy-tools/0.59.4`),
 - changed the permissions (`chmod +x`)

I got this error when compiling in IDE2:

```
fork/exec /Users/a.kitta/Library/Arduino15/packages/teensy/tools/teensy-tools/0.59.4/teensy_size: exec format error

Compilation error: fork/exec /Users/a.kitta/Library/Arduino15/packages/teensy/tools/teensy-tools/0.59.4/teensy_size: exec format error
```

Thank you!

PaulStoffregen, 2024-01-17:
Will build a MacOS copy in several hours.  Right now I have a Windows test machine set up on my workbench.  Need to first wrap up this work with Windows before I can put it away and set up my Mac.

kittaakos, 2024-01-17:
> Will build a MacOS copy in several hours.

You don't need to. Thank you! You've already helped a lot by checking the build from the PR (https://github.com/arduino/arduino-ide/pull/2332#issuecomment-1895699181).

I've set up my Linux env and will go ahead with the already attached binary for Linux.

PaulStoffregen, 2024-01-17:
I built the modified [teensy_size for MacOS](https://www.pjrc.com/tmp/arduino_user_agent_missing/macos/teensy_size).

closed 2024-01-17 by kittaakos

kittaakos, 2024-01-17:
> I built the modified [teensy_size for MacOS](https://www.pjrc.com/tmp/arduino_user_agent_missing/macos/teensy_size).

Thank you! I have verified it on macOS:

Before:

[image]

After:

[image]

PaulStoffregen, 2024-01-18:
Final followup to confirm nightly build now has this issue fixed.

[image]