arduino/arduino-ide #2319: arduino stuck on splash screen if proxy environment variable is configured

mujungh, 2023-12-28:
### Describe the problem

Configuring the http_proxy environment variable under Windows will cause the arduino ide to get stuck on the startup interface. After deleting these two environment variable items, it can start normally, but doing so will cause inconvenience in some areas with network interference.

### To reproduce

Problems occur when configuring http_proxy and https_proxy projects on Windows 10 computers

### Expected behavior

arduino stuck on startup logo interface

### Arduino IDE version

2.2.1

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

tps800, 2024-01-02:
This is only true, if Arduino IDE can reach the internet without accessing a proxy. If it can not access the internet without using a proxy it will just sit there and wait for Godot.
This with or without proxy-variables defined or not. And to make the mess complete: it does not help to give "--proxy-url=http://<yourproxyserver>:<yourproxyport>" at startup. It seems igonored, even if "--help" states it is available.

tps800, 2024-01-02:
Imperfection is nice for "it wont work at all, because it does not start".

DRSDavidSoft, 2024-10-28:
What seems idiotic to me is that in the console, it **clearly** says there is an error, meaning there is no hope of the application _ever_ loading up, so the user has to _stare_ at the loading screen forever.

I wish I could read the mind of developers who code like this. Let's display the splash screen, and **only** stop display it when it reaches a success state. I don't understand why they would not check for startup errors.

Even more infuriating is that the console is not displayed by default. Why not put a small link to display the startup logs at least?

Is this the fault of Theia IDE, or not? I am really interested to know.

PaulStoffregen, 2024-10-28:
To have any chance, please considered editing your message for a softer tone.  I know you're frustrated, but this sort of rhetoric is counterproductive.  Often any entire issue will become locked indefinitely because even just 1 message was "too heated" and then nobody can reply.

Please also understand I don't speak for Arduino.  I've made minor contributions over the years, and I still consider the Arduino developers friends, even though we've not met up in person since before the pandemic.  So I can't say exactly what anyone was thinking, but I do know a bit about Arduino's history.

Several years ago, the decision was made to separate the pre-2.0 software into 2 parts.  All the logic and "real work" would be done by Arduino CLI (command line interface).  Then the IDE would build on top of Arduino CLI.  The basic idea was to move away from the old design where a lot of the logic and mechanics of how the compiler was run ended up inside the Java-based IDE.  There's a lot more to this story, but this reply is already getting long....

Anyway, the point of all this is to understand the software is not one large monolithic design.  It's built as very separate components.  The IDE GUI is also built from multiple independently developed parts.  It's also relatively new, with IDE version 2.0 released only 2 years ago... and a *lot* of development and issues fixed over these last 2 years.  But there are still plenty of issues to be found and fixed.  Along the way, a common theme come up, that some error is reported by Arduino CLI or other other lower level component and the high level GUI which creates the user experience you actually see does not yet have code to gracefully deal with that situation.

Hopefully this explanation helps.  Now if you would please also help, considered editing your message before this issue gets indefinitely locked as "too heated".  Plenty of others, like #58 got locked because of only a small number of angry comments.  Don't let that happen here.

DRSDavidSoft, 2024-10-28:
Wow, @PaulStoffregen! Hi there!

Thanks for the heads-up; I’ve made the changes as you correctly suggested. My apologies for the initial version here. I’m a big fan of your work—from Teensy to your many fantastic Arduino libraries. Hats off to you!

> Often an entire issue will become locked indefinitely because even just 1 message was "too heated" and then nobody can reply.

You’re absolutely right—it’s frustrating when an issue gets locked while the problem persists (like in #58). Sorry to have contributed to that frustration.

> I've made minor contributions over the years

I think you’re being humble! Your contributions to the Arduino ecosystem have been major, and I want to say thank you for all the amazing projects you’ve enabled over the years.

> I still consider the Arduino developers friends, even though we've not met up in person since before the pandemic.

They seem like fantastic people who’ve given the world this amazing open-source platform. It would be an honor to meet them, if I could too.

> the decision was made to separate the pre-2.0 software into 2 parts...

I believe that is an excellent decision made, for two reasons:

1. People who use CLI in their daily workflows can utilize the great tools that `arduino-cli` offers, from installing and managing boards, tools, libraries, to running the compilers and programmers
2. The external utilities and tools that themselves relied on Arduino (for example, the Arduino VS Code extension) don't have to rely using the command line switches for the older Java-based IDE, which in my opinion were kind of finicky 

Plus, with the new CLI built in Go, a compiled language, it’s fast, self-contained, and efficient—a huge win for the community!

> how beneficial that really is I'm not sure, but the Arduino developers definitely think it's a huge positive

I agree—separating the logic from the GUI offers real benefits, as I mentioned. Kudos to the team for moving the core functions into `arduino-cli`.

> There's a lot more to this story, but this reply is already getting long....

A bit off-topic, but I’d love to read more about this. It deserves its own blog post!

> It's built as very separate components. Different parts are even built in completely different programming languages.

Makes sense! The CLI tools are fantastic to use.

> The IDE GUI is also built from multiple independently developed parts. It's also relatively new, with IDE version 2.0 released only 2 years ago... and a *lot* of development and issues fixed over these last 2 years.

While I understand these points, I’m still puzzled that an app developed in the 2020s would suffer from these issues. It almost feels like there’s some oversight somewhere, though I’d be glad to hear if there’s a solid reason behind this design.

> Along the way, a common theme come up, that some error is reported by Arduino CLI or other lower-level component and the high-level GUI which creates the user experience you actually see does not yet have code to gracefully deal with that situation.

Totally agree! While it’s okay if the GUI doesn’t handle every specific error, it should at least recognize when _something_ goes wrong. Many applications address this by showing an error notification. For example, IntelliJ products guide users to view logs when something’s amiss.

Technically speaking, I would like to learn more about the root cause of this issue.

atomicdeploy, 2026-10-14:
Confirmed on **Arduino IDE 2.3.10** (bundled **Arduino CLI 1.5.1**) on **Windows 11 Pro, build 26200** on 2026-10-14.

The IDE remained on the startup/splash screen when the inherited environment contained:

```text
http_proxy=http://localhost:10808
```

Neither `no_proxy` nor `no_grpc_proxy` was defined. The proxy was working: a separate HTTPS HEAD request through it to the Arduino library index returned HTTP 200. This was therefore not simply an unavailable proxy or an unreachable download server.

### Observed startup behavior

The backend and the bundled CLI daemon started and listened on loopback addresses, but the frontend stopped progressing after `attached_shell`, layout restoration began, and plugins were synchronized. The application did not reach `ready`, and no actionable error was shown on the splash screen.

### Verified workaround

I closed that hung IDE instance and relaunched the same executable with only these two process-scoped environment additions, leaving `http_proxy` and the Windows proxy settings unchanged:

```powershell
$env:no_proxy = 'localhost,127.0.0.1,::1'
$env:no_grpc_proxy = 'localhost,127.0.0.1,::1'
& 'C:\Program Files\Arduino IDE\Arduino IDE.exe'
```

The same existing sketch and editor layout loaded successfully. The startup log then recorded:

```text
2026-10-14T10:24:20.607Z root INFO <<< The layout has been successfully restored.
2026-10-14T10:24:21.756Z root INFO Frontend application start: 9746.8 ms [Finished 12.824 s after backend start]
2026-10-14T10:24:23.377Z root INFO Changed application state from 'initialized_layout' to 'ready'.
```

The resulting editor window was responsive.

Inspection of the installed bundled JavaScript shows that its gRPC proxy mapper reads `grpc_proxy`, `https_proxy`, or `http_proxy`, and checks `no_grpc_proxy` first, falling back to `no_proxy`. This supports the explanation that the proxy was being applied to the IDE's internal loopback gRPC connection. No packet capture was performed, and the successful relaunch set both bypass variables together; I have not separately tested each variable on its own.

It would help if the IDE explicitly bypassed external HTTP proxies for its internal loopback communication, or at least displayed a bounded startup failure with an explanation and the applicable bypass workaround instead of remaining indefinitely on the splash screen.

This confirmation is for the installed stable version above; I have not tested a nightly build.

c0mpus3rvef, 2026-10-16:
Hi, having the same issue with Arduino IDE 2.3.10 where the app stuck on splash screen.  am not that technical, appreciate including details as to where the changes you noted under Verified workaround should be done, i.e.  modifying a configuration file and so forth?  
Thank you.

Verified workaround

I closed that hung IDE instance and relaunched the same executable with only these two process-scoped environment additions, leaving http_proxy and the Windows proxy settings unchanged:

$env:no_proxy = 'localhost,127.0.0.1,::1'
$env:no_grpc_proxy = 'localhost,127.0.0.1,::1'
& 'C:\Program Files\Arduino IDE\Arduino IDE.exe'