arduino/arduino-ide #2321: Add Portable Mode (from Theia source)

pscrespo, 2023-12-30:
### Describe the request

Since Theia has implemented portable mode, can it be ported to the Arduino IDE?
References: [https://github.com/eclipse-theia/theia/pull/12690](https://github.com/eclipse-theia/theia/pull/12690)
[https://github.com/eclipse-theia/theia-blueprint/pull/276](https://github.com/eclipse-theia/theia-blueprint/pull/276)

### Describe the current behavior

Current behavior is no portable mode.

### Arduino IDE version

2.x

### Operating system

Windows, Linux

### Operating system version

Windows 11; Debian

### Additional context

I've seen #122, but I can't comment on that one, and thought that this information could be useful for devs.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

linxcow, 2024-01-01:
Lets hope this becomes a reality in 2024 for ide 2.x !

kittaakos, 2024-02-01:
Moved to https://github.com/arduino/arduino-ide/issues/122#issuecomment-1921024673.

Closing as a duplicate

closed 2024-02-01 by kittaakos