arduino/arduino-ide #2328: Need way to copy all Serial Monitor text

JetForMe, 2024-01-08:
### Describe the request

With the current (2.2.1) IDE on macOS, I can't easily select all the Serial Monitor text and copy it to the clipboard. There ought to be a way to do this.

Drag-selecting more than a few lines simply doesn't work.

### Describe the current behavior

No way to select all serial monitor text.

### Arduino IDE version

2.2.1

### Operating system

macOS

### Operating system version

14.2.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-01-08:
Hi . Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1081 and https://github.com/arduino/arduino-ide/issues/812.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the others.

closed 2024-01-08 by per1234

JetForMe, 2024-01-08:
Ah, sorry, I guess I didn't look hard enough. Alas, those are locked and I can't even seem to upvote them. They are also nearly two years old. I sure hope this gets some attention. Even a simple "copy" button that just copies the entire buffer to the clipboard would do.