arduino/arduino-ide #2337: Arduino IDE builtins Apple ARM support

JosefWN, 2024-01-21:
### Describe the request

Even with a boards definition with toolchains etc. fully supporting macOS ARM processors, some IDE builtins do not. I made this umbrella ticket to track support for Apple ARM in Arduino IDE. Feel free to close if you don't find it relevant, figured it could at the very least save people time when compiling their own replacement binaries (which is luckily fairly trivial, so the IDE is on the very cusp of native support).

Builtin's support for Apple ARM:
- [x] `dfu-discovery`
- [x] `mdns-discovery` 
- [x] `serial-discovery`
- [ ] `serial-monitor`: https://github.com/arduino/serial-monitor/issues/69
- [ ] `ctags`: https://github.com/arduino/ctags/issues/20

### Describe the current behavior

The binaries won't execute without Rosetta, leaving the IDE somewhat broken.

### Arduino IDE version

2.2.1

### Operating system

macOS

### Operating system version

14.2.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-01-22:
Thanks for the report @JosefWN. When it comes to the "built-in" tools, fortunately the infrastructure is already in place so that as soon as Apple Silicon builds are added to the [package index](https://downloads.arduino.cc/packages/package_index.json), Arduino IDE will automatically start using those on Apple Silicon machines. No changes will be needed in this codebase and even previous versions of the IDE will have that behavior.

So we don't need to actively track the work for the tool dependencies. This issue will still serve to aggregate a list of the individual issues even now that it is closed.

There are some additional tools that are not distributed via the package index system. Instead, those tools are bundled with the Arduino IDE distribution by the build system. I did a review of those bundled tools and found that a native Apple Silicon variant was bundled with the IDE in all but one case, which was the [**Arduino Firmware Uploader**](https://github.com/arduino/arduino-fwuploader). Fortunately, the build system of that tool was recently updated to produce Apple Silicon builds so the only remaining task was to update the infrastructure here in the Arduino IDE repository to bundle that build when appropriate. That change has now been made, completing all known tasks required in this repository: https://github.com/arduino/arduino-ide/pull/2338

FlorianHeigl, 2024-09-06:
idk, did an install of the ARM .dmg (2.3.2) and yet sitting here trying to fix build issues

the arm builds are running and should in theory everything should be solved as per the comments above.
but it still explodes.
is this to blame on some other software? the path makes me doubt.

```
floh@LaptopvnFlorian 5.8-arduino11 % pwd  
/Users/floh/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11
floh@LaptopvnFlorian 5.8-arduino11 % ls -l ctags 
-rwxr-xr-x@ 1 floh  staff  283868 Sep  5 22:37 ctags
floh@LaptopvnFlorian 5.8-arduino11 % file ctags 
ctags: Mach-O 64-bit executable x86_64
```

per1234, 2026-11-11:
I am assigning this issue to @iabdalkader.