arduino/arduino-ide #2340: Multiple "JavaScript error ocurred in the main process" when launching from a .desktop file

ffpp2003, 2024-01-25:
### Describe the problem

A lot of error windows appear when launching and closing Arduino IDE (either the appimage version or the zip version) on Debian 12, with swaywm. Launching from terminal makes no errors to appear.

This error also occurs on the stable version of the IDE, but with a lot less popup windows

### To reproduce

Use this .desktop file:
```
[Desktop Entry]
Name=Arduino IDE Nightly
Exec=/home/fpp/Aplicaciones/arduino-ide_nightly-20240125_Linux_64bit.AppImage
Terminal=false
Type=Application
Icon=arduino-ide
StartupWMClass=Arduino IDE
X-AppImage-Version=2.2.2-nightly-20240125
Comment=Arduino IDE
Categories=Development;
```
and launch app

### Expected behavior

No errors should appear

### Arduino IDE version

2.2.2-nightly-20240125

### Operating system

Linux

### Operating system version

Debian 12

### Additional context

Picture of the error:
[image]
This error occurs 15 times when launching and 4 times when closing

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

LEKPHET-KUNSUE, 2024-01-27:
test this:
```javascript
console.log("LEKPHET_KUNSUE test!);
``` 
output:
```
LEKPHET_KUNSUE test!
```

ffpp2003, 2024-01-27:
> test this:
> 
> ```js
> console.log("LEKPHET_KUNSUE test!);
> ```
> 
> output:
> 
> ```
> LEKPHET_KUNSUE test!
> ```

Sorry, where should i test those commands?

real-bombinho, 2025-02-01:
Same issue on 

Ubuntu 24.04.1 LTS, 
Arduino IDE Version: 2.3.4, Date: 2024-12-03T10:51:12.539Z, CLI Version: 1.1.1