arduino/arduino-ide #2343: error message:  collect2.exe: fatal error: CreateProcess: No such file or directory

AJ6CU, 2024-01-28:
### Describe the problem

A compilation/build that "was working" started generating the error message: 
"collect2.exe: fatal error: CreateProcess: No such file or directory"

This is for a RaspberryPi Pico, with https://github.com/earlephilhower/arduino-pico, lvgl 8.3.11, espi 

### To reproduce

checkout https://github.com/AJ6CU/KD8CEC-V2.0-for-uBITX/tree/version2.0-tft-ux-experiment. 
Requires:

1.  https://github.com/earlephilhower/arduino-pico
2. lvgl 8.3.11
3. espi  ( https://github.com/earlephilhower/arduino-pico, lvgl 8.3.11, espi )
4. RotaryEncoder 1.5.3 (Mathertel) (must if you are compiling for Pico, optional to replace digital pins for the analogreads) https://github.com/mathertel/RotaryEncoder
5. SparkFun_External_EEPROM (v1.0.12 or better) //https://github.com/sparkfun/SparkFun_External_EEPROM_Arduino_Library

### Expected behavior

obviously a link.  ;-)
Not a lot of reports on googling, closest seems to be:
https://community.nxp.com/t5/MCUXpresso-IDE/Problem-during-linker-phase-quot-collect2-exe-fatal-error/td-p/1586523

See attached verbose compile output, looks like some long command lines there..

[compilationoutput.txt](https://github.com/arduino/arduino-ide/files/14077592/compilationoutput.txt)

### Arduino IDE version

2.2.1 and also tried on nightly available Jan 28, 2024

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

AJ6CU, 2024-01-29:
I found a workaround by creating a static library version of lvgl using the information provided here:

https://forum.arduino.cc/t/precompiling-a-big-library/626128/7

Interestingly, this problem does not occur when I compile for a teensy 4.

kittaakos, 2024-02-20:
> I found a workaround by creating a static library version of lvgl using the information provided here:

Hello, I am glad you found a workaround. I am sure it was a fix outside this issue tracker's scope. We track imperfections and feature requests for Arduino IDE 2.x.

For questions related to your favorite project, please use the forum instead: https://forum.arduino.cc/c/software/ide-2-x/93

 Thanks

closed 2024-02-20 by kittaakos