arduino/arduino-ide #2344: Open dialog run minimized

edyakonov, 2024-01-29:
### Describe the problem

When run ArduinoIDE v2 and choose "File" - "Open..." dialog window open minimized.

https://github.com/arduino/arduino-ide/assets/158043955/736e742b-ab3c-41a6-bc73-00f15141c657

### To reproduce

1. Open Arduino IDE v.2
2. File - Open...

### Expected behavior

Open dialog windows will open minimized.

### Arduino IDE version

Version: 2.2.1 Date: 2023-08-31T14:18:20.829Z CLI Version: 0.34.0  Copyright © 2024 Arduino SA

### Operating system

Linux

### Operating system version

МОС12

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2024-01-30:
From https://github.com/arduino/arduino-ide/issues/2325#issue-2065617796:

> ### Describe the problem
> When I do "File -> Open", the file browser doesn't seem to open, nothing seems to happen, except that Arduino's icon in the Task Manager changes to some "X" icon and has a "+" on the bottom indicating that there's more than one window. When I click on the icon, then the file browser popup comes to the foreground, and so I find out that it did open, but was on the background.
> 
> ### To reproduce
> go to File -> Open
> 
> ### Expected behavior
> Arduino's icon in the task manager shouldn't change to a different icon. The "+" on the bottom might be expected (though most applications do something better: the main window gets greyed out while the file browser popup is visible, and it doesn't count as an additional window). And most importantly, the file browser popup that has been opened as a result of a direct user input, should immediately be visible, focused, and on the foreground.
> 
> ![image](https://private-user-images.githubusercontent.com/13969584/294033664-13ee7ec3-4dff-40fe-8fc4-44ca8fd9abc6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1OTgzMDYsIm5iZiI6MTcwNjU5ODAwNiwicGF0aCI6Ii8xMzk2OTU4NC8yOTQwMzM2NjQtMTNlZTdlYzMtNGRmZi00MGZlLThmYzQtNDRjYThmZDlhYmM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMwVDA3MDAwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQyMGI3YTUyMjFiYTVmZDVlYTFlNWQxOGY3MzkxMzc0NTVkMWRiNDkwMjc0YmM5NDE0MGNiYjgyMGZhNmI2YjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5lgz2oe55FyvYppbsfoitpWOmJBVsd7Kc6mI2eicw9E)
> 
> ### Arduino IDE version
> 2.2.1
> 
> ### Operating system
> Linux
> 
> ### Operating system version
> Manjaro Linux
> 
> ### Additional context
> I did _not_ verify the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds) because the nightly build isn't usable.

kittaakos, 2024-01-30:
I could not reproduce it on my Ubuntu 22.04 VM.

php4fan, 2024-01-30:
Does the Arduino IDE by any chance use Electron?
Visual Studio Code has the exact same issue (or at least it looks pretty much the same): https://github.com/microsoft/vscode/issues/146422

martin-eden, 2024-03-09:
> Does the Arduino IDE by any chance use Electron?

It does. Naive question to ask in source code repository.

https://github.com/arduino/arduino-ide/tree/main/electron-app