arduino/arduino-ide #2352: Working very slow

Ilkhomjon2002, 2024-02-07:
### Describe the problem

Hello .Arduino is working very slow after uploading some sketch to board and serial monitor is not showing clear.Printing some kind of unreadable symbols. I am using it on mac and ubuntu. Both have same issue. I am new to it maybe I am skipping something. Please help me. 

### To reproduce

.

### Expected behavior

.

### Arduino IDE version

2.2.0 , 2.2.1

### Operating system

Linux, macOS

### Operating system version

macOs monterey version 12.6.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-02-08:
Hi @Ilkhomjon2002.

> after uploading some sketch to board

Please provide the sketch in a comment here on the issue thread.

Please also tell us which Arduino board you are using.

per1234, 2025-01-15:
Closing due to lack of response.

@Ilkhomjon2002 if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

I will take a wild guess that this observation:

> Arduino is working very slow

Is using the ambiguous "Arduino" to refer to Arduino IDE, and that it is caused by https://github.com/arduino/arduino-ide/issues/934.

This can be a side effect of the other component of the report:

> Printing some kind of unreadable symbols.

In cases where the serial output from the board is corrupted in some way, even if the sketch code is written to produce reasonable length lines, the line breaks may be lost through the data corruption, resulting in large quantities of content on a single line in Serial Monitor.

The question remains as to what might have caused the corruption of the serial data. It might be a case of Serial Monitor being set to a different baud rate than the sketch is configured for, or a bug in the sketch. I was hoping to be able to identify such a cause by looking at the sketch code, but it was never provided so it is impossible to proceed with the investigation.

closed 2025-01-15 by per1234