arduino/arduino-ide #2357: Unable to connect to boards after version 2.3.0 Update

MarkDerbyshire, 2024-02-07:
### Describe the problem

arvdude  fails to sync when uploading to boards after 2.3.0 update

### To reproduce

Update to 2.3.0 and try to upload

### Expected behavior

Able to upload to boards
[image]

### Arduino IDE version

2.3.0

### Operating system

Windows

### Operating system version

Windows 10 Home - 22H2

### Additional context

Rolled back to 2.2.1 and can now upload sketches to boards

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-02-07:
Hi @MarkDerbyshire. We will need some additional information from you in order to investigate this problem:

Please add a comment here to tell us which board you have selected from the **Tools > Board** menu in Arduino IDE.

---

We also need you to post the full verbose output from an upload attempt:

1. Select **File > Preferences...** from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Uncheck the box next to **Show verbose output during: > `☑` compilation** in the "**Preferences**" dialog.
1. Check the box next to **Show verbose output during > ☐ upload**.
1. Click the "**OK**" button.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for upload to fail.
1. After the upload fails you'll see a button on the "**Upload error**" notification at the bottom right corner of the IDE window: <kbd>**COPY ERROR MESSAGES**</kbd>. Click that button.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a comment here (please use [code fencing](https://guides.github.com/features/mastering-markdown/)).

MarkDerbyshire, 2024-02-07:
Board is the Arduino Nano

```
Sketch uses 5884 bytes (19%) of program storage space. Maximum is 30720 bytes.
Global variables use 518 bytes (25%) of dynamic memory, leaving 1530 bytes for local variables. Maximum is 2048 bytes.
"C:\Users\mark\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\mark\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega328p -carduino "-PCOM3" -b115200 -D "-Uflash:w:C:\Users\mark\AppData\Local\Temp\arduino\sketches\C850D7B61DBDB4E14C76A51EBDE206D3/LED_Control_Multiplex.ino.hex:i"

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\mark\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

Failed uploading: uploading error: exit status 1

```

per1234, 2024-02-07:
OK, it is as I suspected. I think the way Arduino IDE stores the board settings changed since the 2.2.1 release (probably https://github.com/arduino/arduino-ide/pull/2241). This means that all the ["custom board options"](https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options) menus have reset to their default settings.

You are using either a very old official Nano board, or else a cheap Chinese derivative board that still has the pre-2018 bootloader. This means you must select **Tools > Processor > ATmega328P (Old Bootloader)** from the Arduino IDE menus to configure it for uploads to this board. The default value is **Tools > Processor > ATmega328P**, which causes this error when you upload.

Please give that a try and then let us know whether the upload is now successful.

MarkDerbyshire, 2024-02-07:
That worked

Thank You for your time.  Much appreciated

closed 2024-02-07 by MarkDerbyshire