arduino/arduino-ide #2363: Debuging with Arduino IDE 2.3.0 and JLinkGDBServe fails

lboue, 2024-02-10:
### Describe the problem

Debuging with Arduino IDE fails with this error even if I followed this guide:
https://github.com/SiliconLabs/arduino?tab=readme-ov-file#debugger-configuration-file
```
Invalid servertype parameters. The following values are supported: "jlink", "openocd", "stlink", "stutil", "pyocd", "bmp", "pe", "qemu", "external"
```

**debug_custom.json**
```json
{
    "servertype": "jlink",
    "device": "MGM240PB32VNA",
    "interface": "SWD",
    "serverpath": "C:/Program Files/SEGGER/JLink_V794d/JLinkGDBServerCL.exe"
}
```

**Env**
- Arduino IDE version: 2.3.0

[image]

### To reproduce

https://github.com/SiliconLabs/arduino?tab=readme-ov-file#debugger-configuration-file

**debug_custom.json**
```json
{
    "servertype": "jlink",
    "device": "MGM240PB32VNA",
    "interface": "SWD",
    "serverpath": "C:/Program Files/SEGGER/JLink_V794d/JLinkGDBServerCL.exe"
}
```

### Expected behavior

Start debugger.

### Arduino IDE version

2.3.0

### Operating system

Windows

### Operating system version

Windows 11.2312.18.0

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

dankeboy36, 2024-02-10:
Hello, nice report, but not necessarily a bug. Your `debug_custom.json` content looks incorrect if you use IDE 2.3.0+. The JSON must be an array of objects and not an object. See from https://github.com/arduino/arduino-ide/pull/2281#issue-1990271373:

> BREAKING:
>
> A new schema has been introduced for debug_custom.json.

The https://github.com/SiliconLabs/arduino?tab=readme-ov-file#debugger-configuration-file doc needs to be updated.

You must have something like this instead: https://github.com/arduino/arduino-ide/issues/2299#issuecomment-1859699452

```json
[
  {
    "configId": "here comes the FQBN of the board and the programmer. Copy it from the `launch.json`",
    "servertype": "jlink",
    "device": "MGM240PB32VNA",
    "interface": "SWD",
    "serverpath": "C:/Program Files/SEGGER/JLink_V794d/JLinkGDBServerCL.exe"
  }
]
```

lboue, 2024-02-10:
Thanks for the explanation

closed 2024-02-10 by lboue

dankeboy36, 2024-02-10:
Please let me know if you need help getting it working.