arduino/arduino-ide #2365: Checkboxes in Preferences dialog can be changed unintentionally

sterretjeToo, 2024-02-11:
### Describe the problem

If one clicks in the Preferences dialog box on a "line" (vertical position) that contains a checkbox, the checkbox toggles.

### To reproduce

1) Open Preferences
2) Use mouse to click on a "line" with a checkbox (e.g. at the far right.
3) Checkbox will toggle

### Expected behavior

Checkbox should only toggle when clicked on the box itself or the associated text.

### Arduino IDE version

arduino-ide_nightly-20240211_Windows_64bit

### Operating system

Windows

### Operating system version

Windows 10 Home

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2024-02-11:
Thanks for taking the time and letting us know.

> Checkbox should only toggle when clicked on the box itself or the associated text.

I think it works as intended. The preference widget in IDE2 is a re-implementation of the IDE 1.8.x behavior:

https://github.com/arduino/arduino-ide/assets/1405703/9fe9a5f5-fe15-4f46-acd9-ff224fe608f7

sterretjeToo, 2024-02-11:
Yep, you clicked on the "box", not far away from it.

For the three checkboxes at the bottom, click where I did put the red mark

[image]

Note
The issue does not occur for the "show verbose output during ...