arduino/arduino-ide #2368: Debug option absent in 2.3

Kabron287, 2024-02-12:
### Describe the problem

Debug option absent in 2.3 while it present in 2.2.1
[image]

### To reproduce

Just run app.

### Expected behavior

Debug option absent

### Arduino IDE version

2.3

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2024-02-12:
What does the tooltip tell when you hover over the disabled _Debug_ toolbar item?

Kabron287, 2024-02-13:
> What does the tooltip tell when you hover over the disabled _Debug_ toolbar item?

**No programmer selected for Raspberry Pi Pico W**

kittaakos, 2024-02-13:
OK. It is not a bug, but let's check. What happens when you select the programmer for the `Tools` > `Programmer` menu?

A programmer must be selected for the debugger to start in Arduino IDE `2.3.0+`: https://github.com/arduino/arduino-cli/pull/2443.

IDE2 intentionally disables the debug functionality when no board or programmer is selected. See https://github.com/arduino/arduino-ide/pull/2342 for more details. We track a related follow-up at https://github.com/arduino/arduino-ide/issues/2345 that is specifically connected to the custom board options.

You can define a default `programmer` in your `boards.txt` if you're a platform developer. Or you can request this as a feature from your core developers. For more details, see https://github.com/arduino/arduino-cli/pull/2416.

Kabron287, 2024-02-13:
> OK. It is not a bug, but let's check. What happens when you select the programmer from the `Tools` > `Programmer` menu?

Please, show me where `Programmer` is?
[image]

kittaakos, 2024-02-13:
Thank you for providing more information on the problem.

I have opened a follow-up: https://github.com/earlephilhower/arduino-pico/issues/2003

> Arduino IDE users using the current version of `Raspberry Pi RP2040 Boards` platform **must** use version `2.2.1` to enable the debugging.

I decided to leave this issue open for visibility reasons, but this problem cannot be fixed in this repository. I appreciate your understanding.

fpistm, 2024-02-15:
Have the same issue with STM32 core. Honestly I don't understand why a programmer is requested now. It is not very clear how to properly support the new requested configuration.

kittaakos, 2024-02-16:
We created a tester build and made the programmer optional for the debugger. Please help us and let us know how the changes work on your system. The PR is here: https://github.com/arduino/arduino-ide/pull/2371. See the documentation on how to use the tester builds and report back: https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md. We appreciate your help.

fpistm, 2024-02-16:
Hi @kittaakos 
Thanks for the quick feedback and implementation. I will test next monday and give you a feedback.

fpistm, 2024-02-19:
Hi @kittaakos 
I've tested your PR and confirm it works as expected:
[image]

@ubidefeo 
No worry, I know it was experimental. I've also opened some GH issues around the debug config.
My concern was simply to understand why a programmer description was required within the 2.3.0 version. I will investigate the new way the debug should be defined/extended and update the STM32 core to enhance debug user experience.
Finally, I know you (and all Arudino IDE/CLI contributors) do your best to provide good user experience. So thank you all :clap: for all the hard work, I know how it can be hard.

And thanks for fixing the issue with the debug interface within the 2.3.1, as I met it during my first attempts to get the debug working again within the 2.3.0 version and wonder if I made a mistake :wink: .

kittaakos, 2024-02-19:
> We created a tester build and made the programmer optional for the debugger. Please help us and let us know how the changes work on your system. The PR is here: #2371. See the documentation on how [...]

@Kabron287, could you please try out the tester build with the fixes 👆? If they work, we can do a patch release. Thanks for the help!

kittaakos, 2024-02-19:
Linking #103 here.

Kabron287, 2024-02-19:
I'm not so familiar withh github nuances.
I reached that step and what to do next?
[image]

Kabron287, 2024-02-19:
Solved.
And it works now!
[image]

closed 2024-02-19 by kittaakos