arduino/arduino-ide #2377: Opening example with .pde file extension fails silently

per1234, 2024-02-20:
### Describe the problem

If the example for a library has `.pde` file extension, Arduino IDE will silently fail to open it.

### To reproduce

1. Use Library Manager to install version "**1.64.0**" of the "**AccelStepper**" library from Library Manager
1. Select **File > Examples > AccelStepper > AFMotor_ConstantSpeed**

🐛 The example doesn't open. There is no communication of a problem.

### Expected behavior

Arduino IDE opens all valid example sketches, regardless of whether they use a deprecated file extension.

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

11

### Additional context

I see the following in the logs when I try to open the example:

```text
2024-02-20T12:47:15.376Z root ERROR Request cloneExample failed with error: ENOENT: no such file or directory, rename 'C:\Users\per\AppData\Local\Temp\.arduinoIDE-unsaved2024120-60044-18wn560.m01l\AFMotor_ConstantSpeed\AFMotor_ConstantSpeed.ino' -> 'C:\Users\per\AppData\Local\Temp\.arduinoIDE-unsaved2024120-60044-18wn560.m01l\AFMotor_ConstantSpeed\AFMotor_ConstantSpeed.ino' Error: ENOENT: no such file or directory, rename 'C:\Users\per\AppData\Local\Temp\.arduinoIDE-unsaved2024120-60044-18wn560.m01l\AFMotor_ConstantSpeed\AFMotor_ConstantSpeed.ino' -> 'C:\Users\per\AppData\Local\Temp\.arduinoIDE-unsaved2024120-60044-18wn560.m01l\AFMotor_ConstantSpeed\AFMotor_ConstantSpeed.ino'
```

I bisected the regression to 074f65445728cb1950bfe1a01017c8227e43aba6 (does not occur with the build for the previous commit 3eef857b4843d33ca96d36f6e8cc2228f8e386ec)

---

This is a regression of https://github.com/arduino/arduino-ide/issues/518

---

Originally reported by @drf5n at:

https://forum.arduino.cc/t/accelstepper-problems-with-arduino-ide-2-3-1/1225842/10

<a name="workaround"></a>

#### Workaround

1. Select **File > Preferences...** (or **Arduino IDE > Settings...** for macOS users) from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Take note of the path shown in the "**Sketchbook location**" field of the dialog.
1. Click the "**CANCEL**" button.
   The "**Preferences**" dialog will close.
1. Select **File > Open...** from the Arduino IDE menus.
1. Navigate to the `libraries` subfolder of the path you saw in the "**Sketchbook location**" field during step (2) of these instructions.
1. You will see a list of the installation folders of each of the libraries you have installed. Scroll through the list until you find the library for which you want to open an example sketch.
1. Open the folder of the library and then the `examples` subfolder.
1. There you will see a list of the folders of each of the example sketches provided by the library. Open the folder of the example you want to open.
1. Select the `.pde` file in the folder.
1. Click the "**Open**" button.

The example sketch will now open in an Arduino IDE window as usual.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

drf5n, 2024-02-20:
This is broken in IDE 2.3.2.

For example, try opening the:

File/Examples/AccelStepper/Bounce example that refers to this file on my Mac:

    /Users/drf/Documents/Arduino/libraries/AccelStepper/examples/Bounce/Bounce.pde

Using the workaround for a differential test, rename your Random.pde to Random.ino within its library directory and the Arduino IDE will open it as usual.

For example:

    /Users/drf/Documents/Arduino/libraries/AccelStepper/examples/Random/Random.ino

...functions as expected.

Tested on MacOS Sonoma 14.2.1 on Apple Silicon M2 

[image]

No screen shot of the silent failure.

----

I find these other .pde files in my libraries:

```
libraries % find . -name "*.pde" -print
./TinyWireM/examples/Tiny85_Temp_LCD_RTC/Tiny85_Temp_LCD_RTC.pde
./TinyWireM/examples/Tiny85_Temp_LCD/Tiny85_Temp_LCD.pde
./TinyWireM/examples/Tiny85_Temp/Tiny85_Temp.pde
./Adafruit_BMP085_Unified/examples/sensorapi/sensorapi.pde
./AccelStepper/examples/MultipleSteppers/MultipleSteppers.pde
./AccelStepper/examples/AFMotor_MultiStepper/AFMotor_MultiStepper.pde
./AccelStepper/examples/ConstantSpeed/ConstantSpeed.pde
./AccelStepper/examples/Blocking/Blocking.pde
./AccelStepper/examples/ProportionalControl/ProportionalControl.pde
./AccelStepper/examples/AFMotor_ConstantSpeed/AFMotor_ConstantSpeed.pde
./AccelStepper/examples/MotorShield/MotorShield.pde
./AccelStepper/examples/DualMotorShield/DualMotorShield.pde
./AccelStepper/examples/Quickstop/Quickstop.pde
./AccelStepper/examples/MultiStepper/MultiStepper.pde
./AccelStepper/examples/Overshoot/Overshoot.pde
./AccelStepper/examples/Bounce/Bounce.pde
./AFMotorDrawbot/examples/MotorParty/MotorParty.pde
./AFMotorDrawbot/examples/MotorTest/MotorTest.pde
./AFMotorDrawbot/examples/AFMotor_MultiStepper/AFMotor_MultiStepper.pde
./AFMotorDrawbot/examples/AFMotor_ConstantSpeed/AFMotor_ConstantSpeed.pde
./AFMotorDrawbot/examples/StepperTest/StepperTest.pde
./FreqMeasure/examples/Serial_Output/Serial_Output.pde
./FreqMeasure/examples/LCD_Output/LCD_Output.pde
./LiquidCrystal_I2C/examples/HelloWorld/HelloWorld.pde
./LiquidCrystal_I2C/examples/CustomChars/CustomChars.pde
./LiquidCrystal_I2C/examples/SerialDisplay/SerialDisplay.pde
./FrequencyTimer2/examples/Test/Test.pde
./Adafruit_LED_Backpack_Library/examples/custom_character/customgen/customgen.pde
./NXPMotionSense/OrientationVisualiser/OrientationVisualiser.pde
./FreqCount/examples/Serial_Output/Serial_Output.pde
./FreqCount/examples/LCD_Output/LCD_Output.pde
```

testing a few of these, I find they fail as well.

per1234, 2024-03-24:
> This is broken in IDE 2.3.2.

@drf5n that is expected. The fix was made after the time of the 2.3.2 release. Please always check with the latest nightly build of Arduino IDE before reporting problems:

https://www.arduino.cc/en/software#nightly-builds