arduino/arduino-ide #2382: Arduino IDE 2.3.2 does not install newest ESP boards manager (2.0.14)

beta-tester, 2024-02-22:
### Describe the problem

i downloaded Arduino IDE 2.3.2 as AppImage on ubuntu 23.10 linux, made the AppImage file executable and started it.
it downloaded and installed several dependenvies.
if i look to the board manager i can see that esp32 from espressif ist installed, but the newst version in the list is 2.0.11.
[image]

on [espressif / arduino-esp32](https://github.com/espressif/arduino-esp32/releases/tag/2.0.14) there is version 2.0.14 listed as current official release - it was released already on Oct 4, 2023.

the release 2.0.11, that is listed in Arduino IDE 2.3.2 as most actual is from Jul 24, 2023. 

so i am missing several fixes if i can not install the most current official release 2.0.14 from espressif- Arduino IDE 2.3.2 don't offer me the newest official release.

### To reproduce

1. Arduino IDE 2.3.2 as AppImage, make it executable and run it.
2. wait til all dependencieas were installed.
3. restart Arduino IDE 2.3.2
4. open Boards Manager and search for "espressif"
5. Esp32 by Espressif Systems ist listed.
6. open the list box of available version of that package.
7. the list goes just up to 2.0.11 but not higher.
8. from point of view of Arduino IDE 2.3.2, there are no updates for that package available.

### Expected behavior

i expect that the Esp32 package by Espressif Systems will be in sync with the ones at
https://github.com/espressif/arduino-esp32/releases

in this case at this moment it should be 2.0.14 and not 2.0.11

### Arduino IDE version

2.3.2 AppImage

### Operating system

Linux

### Operating system version

Ubuntu 23.10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-02-22:
Hi @beta-tester. Please provide the package index URL you are using for the ESP32 platform.

I'll provide instructions you can follow to do that:

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Click on the "**Additional Boards Manager URLs**" field.
1. Press the <kbd>Ctrl</kbd>+<kbd>A</kbd> keyboard shortcut.
   All the text in the field will be selected.
1. Press the <kbd>Ctrl</kbd>+<kbd>C</kbd> keyboard shortcut.
   This will copy the selected text to the clipboard.
1. Click the "**CANCEL**" button in the "**Preferences**" dialog.
1. Click on the comment field in this issue thread.
1. Press the <kbd>Ctrl</kbd>+<kbd>V</kbd> keyboard shortcut.
   The copied text will be pasted into the comment field.
1. Click the "**Comment**" button to post your comment.

beta-tester, 2024-02-22:
> Hi @beta-tester. Please provide the package index URL you are using for the ESP32 platform.

not sure what you mean. i just downloaded Arduino IDE 2.3.2, started it and installed the out of the box esp32 package (by Esptessif Systems, 2.0.11) that was listed in Arduino IDE 2.3.2 - see screenshot at initial post.
i didn't made any handmade customization.

following your instruction:
the box of "Additional Boards Manager URLs" is empty.

i expected the listed out of the box "esp32 by Espressif Systems" would be up to date - especially Arduino IDE 2.3.2 is only few days old and should have the most recent "esp32 by Espressif Systems" package list included.

where is the "esp32 by Espressif Systems" package in the list comming from?
was it downloaded at the first start of Arduino IDE 2.3.2 or was it already shipped with the AppImage?

after filling "Additional Boards Manager URLs" with: `https://espressif.github.io/arduino-esp32/package_esp32_index.json`
changed the list box of "esp32 by Espressif Systems" package and shows the expected version 2.0.14 as available most recent version.

per1234, 2024-02-22:
>  i just downloaded Arduino IDE 2.3.2, started it and installed the out of the box "esp32 by Espressif Systems" package that was listed in Arduino IDE 2.3.2

Thanks for the clarification. I hadn't realized before now that Arduino had added versions 2.0.10 and 2.0.11 of the 3rd party "esp32" boards platform to the list of platforms that are available by default.

I have reported this problem of the lack of entries for the recent releases of the platform to the repository where that infrastructure is hosted (it is a private repository so I am not able to share a link you could follow). Since the problem is not related to any of the code hosted in this repository, and since there is nothing that can be done to resolve it within this repository, I will close this, but rest assured that we are still tracking it. Thanks for bringing this to our attention!

> where is the "esp32 by Espressif Systems" package in the list comming from?

From here:

https://downloads.arduino.cc/packages/package_index.json

If you are curious about the nature of this file, you can find general documentation about it [**here**](https://arduino.github.io/arduino-cli/latest/package_index_json-specification/).

>after filling "Additional Boards Manager URLs" with: https://espressif.github.io/arduino-esp32/package_esp32_index.json
changed the list box of "esp32 by Espressif Systems" package and shows the expected version 2.0.14 as available most recent version.

Great work finding the solution to getting the up to date version of the platform!

closed 2024-02-22 by per1234

beta-tester, 2024-02-22:
> Thanks for the clarification. I hadn't realized before now that Arduino had added versions 2.0.10 and 2.0.11 of the 3rd party "esp32" boards platform to the list of platforms that are available by [...]

I don't think it is clever to hold an own "merged" paket list that includes versions of 3rd party platforms.
this poses the risk of incompatibility and conflicts when two packet lists lead different versions - which list has the higher priority, the Arduino IDE's  own or the 3rd parties (Espressif's) own list.
in addition, the list must always be kept up to date, which is an additional effort and obviously didn't work out now.

wouldn't it be better to enter the official link to the 3rd party (Espressif) package list in the Preferenzen / "Additional Boards Manager URLs"?
1. the Arduino IDE still shows the esp32 packet lists out of the box.
2. Arduino team is not responsible to keep the list up to date.

problem solved