arduino/arduino-ide #2393: Fix right-click 'Go to Definition' function

Rollpopptery, 2024-02-26:
### Describe the request

Version 2.3.1
Date: 2024-02-15T09:01:25.701Z

No longer opening Tab to the file containing the Variable definition.
Instead small, what appears to be broken , blue-coloured slit-shaped window appearing over the top of the current window, with tiny little viewport and a pokey little scroll bar
I have tried: Re-installing Arduino IDE.
Reverted to version 1.X

### Describe the current behavior

Right click on a variable or object and select 'goto definition'.
You will be presented with the 'half window' from hell.

### Arduino IDE version

Version 2.3.1  Date 2024-02-15T09:01:25:701Z

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-02-26:
Hi @Rollpopptery. Thanks for your report. Please provide a [screenshot](https://www.techradar.com/how-to/how-to-screenshot-on-a-pc) of the Arduino IDE window while this "blue-coloured slit-shaped window" is present in a comment here on this issue thread.

You can take a screenshot by pressing <kbd>**Alt**</kbd>+<kbd>**Print Screen**</kbd> on your keyboard, which will copy the screenshot to your clipboard. You are also welcome to screenshot by any other procedure you prefer.

You can then add the screenshot to a comment here by doing any one of the following:

- Click in the "**Add a comment**" field and press the <kbd>**Ctrl**</kbd>+<kbd>**V**</kbd> keyboard shortcut to [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the screenshot from the clipboard.
- Save the screenshot as an image and then drag and drop it into the "**Add a comment**" field.
- Save the screenshot as an image, click the "**Paste, drop, or click to add files**" at the bottom of the "**Add a comment**" field, and then select the image file.

---

❗ Do not attempt to provide the screenshot via an attachment to an email reply. GitHub does not support attachments in email replies. You must open the issue thread in your web browser to make the comment.

---

per1234, 2025-01-15:
Closing due to lack of response.

@Rollpopptery if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2025-01-15 by per1234