arduino/arduino-ide #2402: Please add option to not display pop-up warnings

rtek1000, 2024-03-13:
### Describe the request

Please add option to not display pop-up warnings

[image]

### Describe the current behavior

After startup, these Pop-up warnings are the things that irritate me the most. The OUTPUT tab doesn't even accept scrolling without having to close the "WONDERFUL" (not to say anything else) pop-up warning.

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Xubuntu - Ubuntu 22.04.3 LTS

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-03-13:
Hi @rtek1000. Thanks for taking the time to submit an issue. I see this is a duplicate of https://github.com/arduino/arduino-ide/issues/2057.

That issue was closed because the requested option is already present.

It is in the [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings). I'll provide instructions for disabling it:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar:  
   [image]
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Preferences**" tab will open in the Arduino IDE main panel.
1. Type `workbench.silentNotifications` in the "**Search Settings**" field of the "**Preferences**" tab.
1. Check the box under the "**Workbench: Silent Notifications**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

rtek1000, 2024-03-14:
Thank you for the return.

So does this mean that the IDE went from a simple program with the function of compiling the sketch, to an AutoCad full of prompt commands?