arduino/arduino-ide #2405: [Arduino IDE 2.x] Cannot see lines longer than 10k characters

rtek1000, 2024-03-19:
### Describe the problem

Cannot see lines longer than 10k characters, in the case of a simple favicon, it is not possible to see the end of the line, although the cursor column counter continues to be incremented.

It may be easier for the user to use an external editor with text wrapping.

### To reproduce

20630 characters:

`<link id="favicon" rel="shortcut icon" type="image/png" href="data:image/png;base64,AAABAAMAEBAAAAEAIABoBAAANgAAACAgAAABACAAKBEAAJ4EAAAwMAAAAQAgAGgmAADGFQAAKAAAABAAAAAgAAAAAQAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAkk3/YKJOD5CiTg+Qok4PkJJOD5CCLg+Qke3/kOJuD5Eijg+Qog3/kIH9/5CSPg+Qok4PkKJOD5CiTg+Qkl3/YKJOD5Cybg/wsm4P8KJN//GSzg/3F66/+Ql+//kpbu/4uR7v+Vmu//h4/u/ztI5P8JId7/Cybg/wsm4P8KJOH5CiTg+Qsm4P8JIt7/RlDl/46U7v87RuP/jpXu/7G28/8lMeD/l5/w/3N56v9wdOr/gIfs/xIo4P8KJeD/CiTh+Qok4PkKJN//SFPl/9DT9/+mrPH/IS/g/8zQ9//5+fz/Ulvm/+7w+//h5Pn/EyLe/zZD4/8gNuH/CiXg/wok4fkJJOD5Gy7g/5CY7/+8wfT/+vv8/0dR5f+nrfL/+/v8/1Nh5//m6Pr/6On6/05X5f/DyPb/KTni/wol3/8KJOH5CCLg+XaB7P8xQOL/Jjjh/1JZ5v8vOOH/5un6//T1/P9NV+T/8/T7/9jc+P9mbef//f39/5Oc7/8IId//CiTh+Qke3/mWoPD/DyDe/2Jt6f+fovD/5Ob6//z8/f+hp/D/fobs//z8/f+gpfD/lp3v//39/f9veOr/CSTf/wok4fkTKeD5jJHt/4mP7f/7/P3//Pz9//X1/P+jp/D/Ulfk/+7v+//19vz/Tlfl/+Pl+v/09Pz/QlDk/yE34f8KJOD5GS3g+YCH7P/Z2vj/9fb8/2526v9NVeT/fILr/+3u+//7+/z/hozs/4+V7v/8/P3/srfz/32E6/9XY+f/CiTg+Qsj4PmWm+//ur/0//z8/P/m5/n/8vP7//z8/f/19vz/iIzs/21y6f/29/z/7vD7/0xU5P/f4vr/OUvk/wok4PkIHt/5jZfv/0FJ4//Jzfb/7O37/9ve+f+jp/H/TlTl/42S7f/29vz/9vf8/2hu6P+pr/L/2d34/w8k3v8KJOH5CSTg+UVS5f9lbun/DR/e/01W5f9kaef/lJjv/+Hi+f/8/P3/7u/7/2lu6P+Qk+7/+Pn8/2Zw6f8JI9//CiTh+Qok4PkJId7/h4/u/zQ/4v/P0/f//f39//39/f/z9Pz/s7fz/0xR5P+orfH/+Pj8/5GX7v8MIt//Cybg/wok4fkKJOD5Cybg/xUq4P8iN+H/MD/i/5ag8P9ye+v/RE/k/3uA6//e4Pn/2dz4/2Zv6f8LIt//Cybg/wsm4P8KJOH5CiTg+Qsm4P8KJeD/CiXf/wol3/8IIN7/CSPf/yY64v9cZuj/PU3k/xEl3v8JI9//Cybg/wsm4P8LJuD/CiTh+Qkk3/YJJOD5CSTg+Qkk4PkJJOD5CSTg+Qkk4PkJJOD5CSPg+Qkj4PkJJOD5CSTg+Qkk4PkJJOD5CSTg+Qkl3/YAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKAAAACAAAABAAAAAAQAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAkk3+4JJOD0CSTg9Akk4PQJJOD0CSTg9Akk4PQJJOD0CSTg9Akk4PQJJOD0CSTg9Akk4PQJJOD0BiHf9AUe3/QGHd/0BR/f9AYh4PQJJOD0CSTg9Akk4PQJJOD0CSTg9Akk4PQJJOD0CSTg9Akk4PQJJOD0CSTg9Agk4PQKJuDuCSPg+Asm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/CiXg/wok3/8HIN//CBfd/wsb3v8QKeD/HTLh/yI24f8bMeH/Dybf/wsZ3f8HGN3/ByHf/wok3/8KJeD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/CSTg/wwm4PgJJOD0Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wkk3/8HHt7/Ex/e/zdG5P9we+v/rLHz/8vQ9v/Kz/f/yM73/8vQ9//Lz/b/pKrx/2hz6v8yP+P/Dx7e/wce3v8KJN//Cybg/wsm4P8LJuD/Cybg/wsm4P8JJOD/DCbh9Akk4PQLJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8IH97/CRve/01V5v+mrPP/19n5/6mw8/99gez/Zmjn/09U5f9IT+X/UVfm/2lq6P9/hu3/r7b0/9PW+P+Yn/D/P0fk/woa3f8IIt//Cybg/wsm4P8LJuD/Cybg/wkk4P8MJuH0CSTg9Asm4P8LJuD/Cybg/wsm4P8KJd//Bx3e/yYv4P+Vm/D/x8r3/46T7/85ReP/LDbg/11o6P+Tme//YWvo/wgU3P8PHt7/NUTk/1Fg5/88QuL/SU7k/5md8P/Lzvf/hYvt/xwm3/8GHt7/CiXg/wsm4P8LJuD/CSTg/wwm4fQJJOD0Cybg/wsm4P8LJuD/CiXf/wUZ3f86QeP/sbb0/6Op8v88ROL/BRPc/yMw4P+8wPX/9vb8//v7/f/X2vj/Iy/g/1pj5//g4vr/9/j9/9vd+f9vd+v/ERXd/0xT5v+4vvX/qK3y/y894/8JId//CiXg/wsm4P8JJOD/DCbh9Akk4PQLJuD/Cybg/wom4P8HHd7/PEPj/8LG9v/Eyfb/fIPs/0ZT5f8IG97/Ii/h/8jN9//9/f3//f39//P0/P9hauj/QUvj/+jr+//9/f3//f39/73E9f8cI9//BRfd/yUt4P+iqPH/ZnTq/wce3v8KJeD/Cybg/wkk4P8MJuH0CSTg9Asm4P8LJuD/CB/e/ygy4f+2vfT/0dT3/+nq+//29vz/4eT6/1Vg5/8GFNz/dYHs//f4/f/9/f3/+Pn8/4yQ7v8aKuD/2d35//39/f/9/f3/ztP3/ycv4P8FH97/Bxrd/wwd3v8LIt//CiXf/wsm4P8LJuD/CSTg/wwm4fQJJOD0Cybg/wkj3/8KHN7/m6Px/6ev8/+ZovD/+Pj9//z8/f/+/v3/mKLx/wwU3f9WX+f/7e78//39/f/6+v3/m6Pw/wwj3//L0Pf//f39//39/f/V2Pj/LzXg/1Fi6P+hqvL/font/xgl3/8JIt//Cybg/wsm4P8JJOD/DCbh9Akk4PQKJt//Bhze/1Vf5//Izff/OELi/2946v/x8vz//f39//T1/P9zfuz/BxLc/2h06v/09Pz//f39//n5/P+Rl+//FSff/9XZ+P/9/f3//f39/9HV+P8xNd//h5Hu//z8/f/z9Pz/eoPs/wob3v8KJd//Cybg/wkk4P8MJuH0CSTg9Aok3/8WIt//rLXz/4aP7v8HFNz/FCbf/3eE7P+utvT/fYjt/xcl3/8WHt7/rrbz//39/f/9/f3/9fX8/3B36f8xPOH/5ef6//39/f/9/f3/w8n2/ysw3f+XoPD//v79//39/f/h5Pr/NUTj/wYe3v8LJuD/CSTg/wwm4fQJJOD0Bh7f/0BR5v/Z3Pj/MT/j/wYf3v8IIt//CBbd/xMW3f8KE9z/FBzd/3yE7P/z9Pz//f39//39/f/k5/r/Mj/i/2Nq5//w8fv//f39//39/f+lr/P/ICjd/7i99P/9/f3//f39//T1/P9CVOX/BRve/wsm4P8JJOD/DCbh9Akk4PQJFt3/f4rt/6Su8v8SH97/CCDe/wcb3v8oM+H/Rkvl/2Jn6P+orvP/8vP8//39/f/9/f3//f39/5We7/8aIt3/p7Dz//v8/f/9/f3/9fb9/3Z96v8lNOH/4eP5//39/f/9/f3/2Nv4/yM14f8GIt//Cybg/wkk4P8MJuH0ByPg9A0g3/++wvX/eobs/wQY3f8gK+D/iI/v/9PX+P/l5/r/8fL8//z9/f/9/f3//f39//v8/f/Fy/b/Lzjg/05a5f/r7fv//f39//39/f/f4/r/NkHh/2Fr6P/09fz//f39//39/f+utfP/FB7e/wkj3/8LJuD/CSTg/wwm4fQFH9/0GTDh/83S9/9XXub/FyDe/6yx8//4+f3//f39//39/f/9/f3//f39//39/f/9/f3/x8r2/0BE4v8tMt//xcr2//39/f/9/f3//Pz9/5+n8P8UH93/r7Tz//v8/f/9/f3/+fn9/3aD7P8HGd3/ByDe/wol3/8JJOD/DCbh9AYd3/QqO+L/yM33/0RK4v9qdOn/9/j8//39/f/9/f3//f39//r6/f/19vz/5eb6/5if8P8yOOD/LDDf/7K48//6+v3//f39//39/f/g4/n/OkLi/05U5f/n6fv//f39//39/f/d4Pn/PEXh/1Bf5/9od+v/DCDf/wgj3/8MJuH0Bhze9C8/4//Dyvb/OkPj/7K18//8/P3//f39//b3/P/Gyfb/nKHw/3Z76/84Q+L/GiDd/0xW5f/Bxfb/+vr9//39/f/9/f3/8/T8/3aB6/8eKN3/sLf0//z8/f/9/f3/+/v9/6Ws8f8UH93/s7fz/6Ks8v8QHd7/CCPf/wwm4fQGHd70Kjvi/8TK9v8/SOT/u7z0//z8/f/9/f3/5Of6/0lR5f8PHt//Kzbg/11j5v+jqfL/6Or7//39/f/9/f3//f39//f3/P+ZoPD/GR3c/3mB7P/29/z//f39//39/f/p6/v/QEzj/0hR4//o6vv/nafx/w0d3v8II9//DCbh9AUf3/QYL+D/y9D3/1ld5v+gpfD/+/v9//39/f/5+vz/1tj3/8vN9v/h5Pr/7/D7//v7/f/9/f3//f39//39/f/z9Pz/mZ/w/yIl3f9cY+f/4+b6//z9/f/9/f3/+fr9/42W7v8eJd3/p6/z//j5/f+DkO7/BR3e/wgj3/8MJuH0ByPg9Asg3/+9wvT/d33r/1pn6P/29/z//f39//39/f/9/f3//f39//39/f/9/f3//f39//39/f/8/P3/3+L5/3yD6/8aHdz/WmDn/93h+v/9/f3//f39//z8/f/Hzff/KjHf/1xk5//w8fz/7vD8/1pj6P8FHd7/CCTg/wwm4fQJJOD0CRfd/32I7f+krvL/JCze/5qg8P/z9Pz//f39//39/f/9/f3//f39//39/f/29v3/4eP6/6Gm8f88QuP/ICfd/3N76//i5Pr//f39//39/f/9/f3/3uH5/0hS5P8uOeD/09f4//7+/f/W2/j/Jzbh/wge3v8JJOD/DCbh9Akk4PQGHt7/P03l/9fb+P8yO+L/Fx7e/3R+7P/Axvb/3uD5/9rd+f/Jzff/rLTz/3uB7P88Q+L/FB3d/0lM5P+tsvP/9PX8//39/f/9/f3//f39/+Tm+v9dZOf/HyTd/6yy8//6+/3/+/v9/5eg8P8HGt3/CSTf/wkk4P8MJuH0CSTg9Aol3/8VId//qbHz/4iR7v8IFdz/Bxze/x8p4P8zOeH/NDjg/y0y3v8jKN7/JS/h/1xj5v+pq/L/5Ob6//z8/f/9/f3//f39//z8/f/e4fn/XmTn/x8h3f+XnfD/9vf8//39/f/h5Pn/NUPk/wYd3v8LJuD/CSTg/wwm4fQJJOD0CiXg/wYc3v9TXOf/zNH4/zlE4/8FGd3/CSHe/1Fi6P98iO3/kJjv/7K18//c3Pn/9PT8//v7/f/9/f3//f39//39/f/4+f3/x8z3/0tS5P8fI93/lpzw//T1/P/9/f3/8/T8/3eA7P8MGt7/CiXg/wsm4P8JJOD/DCbh9Akk4PQLJuD/CSPf/w0d3f+WnvD/q7Hz/x4n3/8NHt3/s7v0///+/f/+/v3//f39//39/f/9/f3//f39//39/f/6+v3/6er7/5CV7v8qL9//Lzjg/6mv8//29/z//f39//b2/P+covD/FiLe/wkj3/8LJuD/Cybg/wkk4P8MJuH0CSTg9Asm4P8KJuD/ByDe/yQu4P+6wfX/k5rv/xIe3v+UnPD/+Pn9//39/f/9/f3//f39//39/f/4+f3/3eD5/6Wr8f9ETuP/HyTd/1pf5//R1Pj/+fr8//z8/f/z8/z/m6Dw/xom3/8HH9//Cibg/wsm4P8LJuD/CSTg/wwm4fQJJOD0Cybg/wsm4P8LJuD/Bx3e/zpJ5P9seuv/DCDe/yEt4P+Lku7/5uj7//T1/P/b3fn/srj0/3qE7f89ROH/FyHd/0VL4/+mq/L/7e77//7+/f/7+/3/3uD5/3l/7P8UId7/Bx/f/wsm4P8LJuD/Cybg/wsm4P8JJOD/DCbh9Akk4PQLJuD/Cybg/wsm4P8KJeD/CSDe/wce3v8KJeD/CSLe/w0e3v87TeX/Rlfm/yY44f8WH97/Cxnd/1Fd5/+sr/L/5+j7//n6/f/v8Pz/2Nv5/5ac8P81ROP/DRre/wgi3/8KJuD/Cybg/wsm4P8LJuD/Cybg/wkk4P8MJuH0CSTg9Asm4P8LJuD/Cybg/wsm4P8KJeD/CiXf/wsm4P8LJuD/CSXf/wYc3v8FGt3/ByHf/wgi3/8KId7/doTt/6qz9P+jrfP/i5bv/15m6P8tOOH/Bxvd/wYd3v8KJd//Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/CSTg/wwm4fQJJOD0Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cibg/wol3/8OIN7/Ex7e/xAd3v8HHd7/BR3e/wcf3v8JJN//Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8JJOD/DCbh9Akl4PgLJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wok3/8KJN//CiTf/wok3/8KJeD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wkl4P8MJ+D4CCPf7gkk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0CCTg9Agk4PQIJOD0ByPg9Aol4O4AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKAAAADAAAABgAAAAAQAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgj4ecJJODvCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCCTg8gci4PIHId/yByHf8gcg3/IHId/yByHf8gci4PIII+DyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8gkk4PIJJODyCSTg8ggk4PIHJOHvCyfi5wkj4PYLJeD6CiXg+wol4PsKJeD7CiXg+wol4PsKJeD7CiXg+wol4PsKJeD7CiXg+wol4PsKJeD7CiXg+wol4PsKJOD7CSLf+wkg3/sIH9/7Bh/f+wMe3/sEHN77Bhve+wca3vsGG977BBze+wMd3/sFH9/7CB/f+wkg3/sJIt/7CiTf+wol4PsKJeD7CiXg+wol4PsKJeD7CiXg+wol4PsKJeD7CiXg+wol4PsKJeD7CiXg+wok4PsJJOD6DSfh9wkk3/MKJd/8Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/Cybg/wsm4P8LJuD/CiXg/wkk3/8II [... cut ...] AAAAAAAA==">`

### Expected behavior

- See lines longer than 10k characters.

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Xubuntu 22.04.3 LTS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2024-03-20:
@rtek1000 
I'm trying to understand what's the reason to make such an issue relevant.
If you need to have this data files compiled into your firmware I advice you put them in separate `.h` files  and include them, but ignore them as they're not even human readable/editable files.
You're certainly not editing PNGs in the Arduino IDE

Also... respect to you if you're able to decode base64 💪🏼

rtek1000, 2024-03-20:
> @rtek1000 I'm trying to understand what's the reason to make such an issue relevant. If you need to have this data files compiled into your firmware I advice you put them in separate `.h` files and [...]

I really don't think it's relevant, as it is very common for Arduino IDE users to use other text editors, usually Notepad++. But it would be interesting to see if VS Code can display beyond this limit, as it is becoming a clone, and it may be interesting to continue on this path, as long as it can offer better performance.

The idea of putting it in an h file is adequate. As for decoding, as you can see, it is an image that the browser requests when opening the page, called [favicon.ico](https://www.w3schools.com/html/html_favicon.asp).

`<link id="favicon" rel="shortcut icon"`

> A favicon is a small image displayed next to the page title in the browser tab.

I used Base64 in a [project](https://github.com/rtek1000/Datalogger_2039), but it was a Kazuki Ota library, I just made the header to operate in C, due to the project being started in C on the STM32CubeIDE, and it's not secure either, just unreadable. Looking at the source code, it doesn't seem to be very complicated, here's an explanation, in case anyone is curious:

> The Base64 decode algorithm converts plain text into original data. Technically, it can be said that it converts six-bit bytes into eight-bit bytes.

https://base64.guru/learn/base64-algorithm/decode

For example, "Hello" becomes "[SGVsbG8=](https://www.base64encode.org/)"

Thank you.

dankeboy36, 2024-03-20:
> ### Expected behavior
> * See lines longer than 10k characters.

As a workaround, you can use the `editor.wordWrap` [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings). Set it to `"on"`:

```jsonc
{
  "editor.wordWrap": "on"
}
```

Or use the <kbd>Alt</kbd>+<kbd>Z</kbd> (<kbd>⌥</kbd>+<kbd>Z</kbd> on macOS) keybinding.

You can also use the `View: Toggle Word Wrap` command from the _Command Palette_ to switch between the four modes.

https://github.com/arduino/arduino-ide/assets/111981763/2cc4e75b-8655-434f-a791-20776b19f409

rtek1000, 2024-03-20:
> You can also use the `View: Toggle Word Wrap` command from the _Command Palette_ to switch between the four modes.

Yes, I had considered that, but I found it easier to use an external editor than to look for information about auxiliary configurations in the IDE.

In VS Code a button appears "Show more 10.4KB". I hope the Arduino IDE doesn't have the same bugs as VS code too.

There are 7588 open issues:
https://github.com/microsoft/vscode/issues

martin-eden, 2024-10-27:
> There are 7588 open issues:
> https://github.com/microsoft/vscode/issues

`9762` for today.

I wonder which project having faster issues acceleration. But for me they both looks doomed.