arduino/arduino-ide #2412: IDE 2.3 crashes with "GPU process isn't usable. Goodbye"

jschwender, 2024-04-09:
### Describe the problem

The IDE does not start. When started from command line the error is: GPU process isn't usable. Goodbye.
Latest version that starts properly is 2.2.1, with current 2.3.2 it crashes, no matter if i tried --no-sandbox or --in-process-gpu or --disable-gpu-sandbox. Tried under X11 and Wayland, with same result.

[1620968:0410/002041.542818:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
2024-04-09T22:20:41.741Z root INFO Deploy plugins list: 82.7 ms [Finished 1.528 s after backend start]
[1620968:0410/002041.829197:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[1620968:0410/002042.102192:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[1620968:0410/002042.374797:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[1620968:0410/002042.498552:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[1620968:0410/002042.624619:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[1620968:0410/002042.746267:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[1620968:0410/002042.746285:FATAL:gpu_data_manager_impl_private.cc(448)] GPU process isn't usable. Goodbye.
Trace/Breakpoint ausgelöst (Speicherabzug geschrieben)

### To reproduce

Ubuntu 23.10, nvidia GPU. Download and install IDE into a user accessible directory and start it from there.

### Expected behavior

IDE starts up.

### Arduino IDE version

2.3.2

### Operating system

Linux

### Operating system version

Ubuntu 23.10

### Additional context

NVIDIA Corporation TU117 [GeForce GTX 1650] (rev a1), run with nvidia-dkms-550. Driver version 550.54.14.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-04-10:
Hi @jschwender. Please try starting Arduino IDE from the command line and add the `--disable-gpu` flag to the invocation:

```text
./arduino-ide --disable-gpu
```

Add a comment here to let us know whether it was able to start up and work as expected when you did that.

jschwender, 2024-04-10:
Version 2.3.2, with --disable-gpu:

[243919:0410/085158.291815:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[243919:0410/085158.432116:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[243919:0410/085158.432144:FATAL:gpu_data_manager_impl_private.cc(448)] GPU process isn't usable. Goodbye.
Trace/Breakpoint ausgelöst (Speicherabzug geschrieben)

jschwender, 2024-04-10:
the flatpack edition also does not work, same result:
arduino-ide_2.3.2_Linux_64bit.AppImage
…
[252216:0410/085427.819809:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[252216:0410/085427.944788:ERROR:gpu_process_host.cc(991)] GPU process exited unexpectedly: exit_code=256
[252216:0410/085427.944808:FATAL:gpu_data_manager_impl_private.cc(448)] GPU process isn't usable. Goodbye.
Trace/Breakpoint ausgelöst (Speicherabzug geschrieben)

same result when running  with --disable-gpu

jschwender, 2024-04-15:
Can anyone confirm my observations? Is the observed behavior intentional or is it a bug?

martykube, 2024-04-15:
I had the same result with arduino-ide_2.3.2_Linux_64bit.AppImage, Ubuntu 22.04.4 LTS, GPU:
VENDOR= 0x0000 [Google Inc. (Intel)], DEVICE=0x0000 [ANGLE (Intel, Mesa Intel(R) UHD Graphics (CML GT2), OpenGL 4.6 (Core Profile) Mesa 23.0.4-0ubuntu1~22.04.1)], DRIVER_VENDOR=Mesa, DRIVER_VERSION=23.0.4 *ACTIVE*

jschwender, 2024-04-18:
> I had the same result with arduino-ide_2.3.2_Linux_64bit.AppImage, Ubuntu 22.04.4 LTS, GPU: VENDOR= 0x0000 [Google Inc. (Intel)], DEVICE=0x0000 [ANGLE (Intel, Mesa Intel(R) UHD Graphics (CML GT2), [...]

Did you use X11 or Wayland?

jschwender, 2024-04-18:
I noticed that the most recent version 2.2.1, despite starting correctly throws the same  error about GPU. But it seems too successfully disable the use of it:

[819366:0418/131443.082789:ERROR:gpu_process_host.cc(953)] GPU process exited unexpectedly: exit_code=256
[819366:0418/131443.381627:ERROR:gpu_process_host.cc(953)] GPU process exited unexpectedly: exit_code=256
[819674:0418/131443.390567:ERROR:gpu_init.cc(523)] Passthrough is not supported, **GL is disabled**, ANGLE is 

I ran another test on a pc with AMD GPU where it starts properly.

martykube, 2024-04-18:
> Did you use X11 or Wayland?

Wayland

martykube, 2024-04-18:
> Wayland

I rolled back to arduino-ide_2.2.1_Linux_64bit.AppImage which works just fine.

BohdanTkachenko, 2024-05-07:
This seems to be an issue only when trying to use Wayland. I rolled back to 2.2.1 and it worked for me too. Here is a one-liner for Flatpak:

```
sudo flatpak update --commit=4f8cc54c47c5c5f99fc59130e767d4200e74f468558cfaf1af37666d4c5e0e6e cc.arduino.IDE2
```

BohdanTkachenko, 2024-05-07:
I checked on another machine with AMD GPU -- it works as expected, so the issue only happens on NVIDIA cards.

I also checked all versions between 2.2.1 and 2.3.2 and it seems that the first broken version is 2.3.0.

Taronyu, 2024-08-28:
I have an AMD 5700G APU with integrated graphics and it crashes too on Fedora 40. So not only Nvidia. Passing `--disable-gpu` fixes it for me as a workaround