arduino/arduino-ide #2413: Digispark ATtiny85 flashing is not possible in IDE 2.3.2 Older IDE 1.8.19 does well

woodlist, 2024-04-09:
### Describe the problem

Upon compilation immediately appears `Failed uploading: uploading error: exit status 0xc0000135` instead of "Plug your device in 60 sec."

### To reproduce

"Blink" sketch

### Expected behavior

As it was on IDE 1.8.19

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

Windows 11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2024-04-10:
Hi @woodlist. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

woodlist, 2024-04-11:
> Hi @woodlist. Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:
> 
> https://forum.arduino.cc/

You were absolutely right. I have accidentally discovered the root of trouble and it is not related to the Arduino IDE. I would like to report on the troubleshooting here, because it could be useful for others. The Digispark has been connected to PC via USB 2.0 hub and the trouble gone. This board is not USB 3.0 friendly.