arduino/arduino-ide #2414: Function list view

Shumatic, 2024-04-11:
### Describe the request

I would like a view where i can see alle the function that is definde in the sketch
It could like the view in Notepad++ v.8.6.4   menu - view - Function list
This provide a rapid way to goto a function.
Preferrebly it should work across Tabs
[image]

### Describe the current behavior

N/A

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2024-04-11:
Hi @Shumatic. Arduino IDE already has this feature. It is called the "Outline View".

You will find instructions for enabling the Outline View in the previous requests for the feature: https://github.com/arduino/arduino-ide/issues/237 and https://github.com/arduino/arduino-ide/issues/1157

closed 2024-04-11 by per1234

Shumatic, 2024-04-11:
I am so sorry for not investigating properly ..... :-)