arduino/arduino-ide #2418: Hanging Downloading index: package_index.tar.bz2

niiiksh, 2024-04-17:
### Describe the problem

When I start Arduino IDE it always tries to download such files. I cannot compile anything when this process working and I cannot stop it. When I restart IDE it runs this download again. How to disable this useless downloading? It should be high priority I cannot work until download finishes (about 20 min).

### To reproduce

Run Arduino IDE, Windows 11 Pro last release.

### Expected behavior

IDE let me compile projects and not hang.

### Arduino IDE version

2.3.2

### Operating system

Windows

### Operating system version

22621.3447

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

niiiksh, 2024-05-03:
Will anybody fix it? IDE hanging with NOT CONNECTED and downloading file for 20 minutes with fast internet!

DuckPaddle, 2024-05-13:
I have a similar problem.

HJSK, 2024-06-02:
I have also the exact same problem Downloading index: package_index.tar.bz2
After about 1.5 - 2 min I can work with IDE. 
Rest of the day it does not come back even on restarting IDE or even rebooting PC (win 10 and win 11 PC)

per1234, 2024-08-01:
Hi @niiiksh. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/2347.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2024-08-01 by per1234

jsketcham, 2024-10-04:
On the Mac, deleting the folder /Users/yourUserNameHere/Library/Application Support/arduino-ide, worked.

AndKe, 2024-10-08:
same issue in 2.3.3 ...

cz241239, 2024-10-12:
Close tab "Serial monitor".

miro-ux, 2024-10-21:
> Close tab "Serial monitor".

This need to be on more places!

Worked for fixing hangs on Downloading index: package_esp32_indexjson and package_index.tar.bz2

Sovenok-Hacker, 2024-10-21:
> Close tab "Serial monitor".

Didn't help me. 2.3.3 Linux

aserebryakov, 2024-10-21:
> Close tab "Serial monitor".

Just... How?.. Why?..

miro-ux, 2024-10-21:
> > Close tab "Serial monitor".
> 
> Didn't help me. 2.3.3 Linux

Try starting the IDE with no internet connection. 🤷🏻‍♂️

aserebryakov, 2024-10-21:
> > > Close tab "Serial monitor".
> > 
> > 
> > Just... How?.. Why?..
> 
> When you close the monitor, the download continues. It provides a solution before the error is fixed

I meant connection between these things isn't quite obvious.